bengtmartensson/IrScrutinizer #254: Capturing hw -> Command in sub-process impossible to Stop

irplus-remote, 2018-09-13:
Sorry for bothering :D Another issue: When i clock on "STOP" Irscrutinizer freezes.

[image]

bengtmartensson, 2018-09-13:
It works for me :-). To troubleshoot many more details are needed. Most likely it is a problem with your system.

Actually, there are so much that can go wrong here -- even though the code as such is correct, at least so I believe. I strongly consider removing it.  And the /dev/lirc interface has gone *so much better*. Any reason you are not using that instead?

irplus-remote, 2018-09-13:
You're right dev/lirc works, i wasnt aware of that, then you might remove it and i guess it's fine

bengtmartensson, 2018-09-14:
The /dev/lirc interface is doing things right. The mode2 stuff an ugly kludge...

To the original problem: since it works for me, and you do not give details for reproducing the problem, there is not much I can do. Most likely it is a problem of your JVM which may have a flawed Process.close or Process.destroy, but that is of course also a guess.

For there reasons. closing.

closed 2018-09-14 by bengtmartensson

irplus-remote, 2018-09-14:
Btw: im using this nice jvm here: https://www.bell-sw.com/java.html  on RPI

I may suggest to remove that mode completely then if /dev/lirc works? Then it will cause no confusion. Or hide it under some advanced checkbox

bengtmartensson, 2018-09-14:
There is already an (undocumented) command line option `--experimental`, that I may use...