bengtmartensson/IrScrutinizer #256: Suggestion for Pronto Scrutinize, ignore burst-pair count number or auto-fix it

irplus-remote, 2018-09-17:
I think this worked some time ago, i could just add sequences to pronto and read it correctly (it would autofix the burst pairs i think) 

Example, paste this into scrutinize - > OK
```
0000 006C **0022 0002** 015B 00AD 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 06FB 015B 0057 0016 0E6C
```

Error (not explained), just added one repeat-sequence
[image]

```
0000 006C 0022 0002 015B 00AD 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 06FB 015B 0057 0016 0E6C 015B 0057 0016 0E6C
```
So to parse, i would need to add 2burstpairs either to 0022 or 0002.

It would be cool having either an option which let me ignore the burst pair count or just have irscrutinizer auto-fix it.

bengtmartensson, 2018-09-18:
Thank you very much for the suggestion.

There is a problem with "auto-fixing", since if the length of the intro, the length of the repeat, and the data are not consistent, which one is wrong? The both length can only in addition be compared to the actual length of the data, too.

The idea to "forgive" formal error (differently put, by semantic or syntactic errors, *guess* what the user meant) is about as old as computing itself. Almost always, this leads to more problems than it solves; the "new" problems often quite hard to understand, in contrast to the "old" one.

Having said that, I absolutely agree that the error messages (for example the one you showed above) are not very helpful. One improvement is already planned, #233.

Of course it may be possible to have a user inquiry like "inconsistents, lengths detected, Want to fix

* intro length, or
* repeat length?"

but that is meaningful only after the IrpTransmogrifier migration #212.

irplus-remote, 2018-09-18:
Actually the reason why this comes up in the first place is when i want to add some repeats to a pronto code. While of course i could just calculate the new number its still one step of more work. So some "Add repeat sequence" Button would also be a help (this is a new request basically :D )

I like the idea of asking the user if he wants to fix intro or repeat. This would be perfectly fine instead of showing the error.

bengtmartensson, 2019-08-24:
> Of course it may be possible to have a user inquiry like "inconsistents, lengths detected, Want to fix
>
>    intro length, or
>    repeat length?"

Third alternative: Discard excess data (if more data than into-length + repeat-length).

bengtmartensson, 2019-08-29:
[The commit to IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier/commit/22684a860e485427100bbb3a3be1869293c41fdb)  (and a previous one) introduces loose parsing of Pronto signals. If the lengths do not add up, actual data length (provided it is even!) is taken to be the length of the intro length and repeat length is set to 0. A warning is issued. This commit is now in the CI build. 

@irplus-remote : whaddya say?

irplus-remote, 2019-08-29:
Sounds good to me :) (Should be like it was once), Issuing a warning is okay

bengtmartensson, 2019-08-29:
Ok, closing as fixed.

> (Should be like it was once),

Don't understand, it has never accepted erroneous signals...

closed 2019-08-29 by bengtmartensson