bengtmartensson/IrScrutinizer #259: Can't scrutinize NEC2-f16, but it was recognized?

irplus-remote, 2018-10-20:
[image]

Stumbled across this problem.. Anything i can do ?

bengtmartensson, 2018-10-20:
The rendering engine does not know the said protocol (although the decoder may). You can add it to IrpProtocols.ini by adding these lines to it, for example at the end:

    [protocol]
    name=NEC2-f16
    irp={38.4k,564}<1,-1|1,-3>(16,-8,D:8,S:8,F:16,1,^108m)* [D:0..255,S:0..255=255-D,F:0..65535]

Since I am concentrating my efforts on IrScrutinizer version 2 (including IrpTransmogrifier) #212, and you can easily fix the problem individually (as above), I do not intend to check it in for IrScrutinizer 1.*.* /IrpMaster.

Thanx for the report anyhow; will go into IrScrutinizer 2/IrpTransmogrifier.

bengtmartensson, 2018-10-22:
Closing for reasons given above. Feel free to further comment though.

closed 2018-10-22 by bengtmartensson

irplus-remote, 2019-08-28:
Hey, now with IrScrutinizer2 the NEC2-f16 is not recognized at all You can try this to test:

```xml
# Please take the time to finish this file as described in
# https://sourceforge.net/p/lirc-remotes/wiki/Checklist/
# and make it available to others by sending it to
# <lirc@bartelmus.de>
#
# This config file was automatically generated
# using lirc-0.9.4c(default) on Tue Aug 27 18:54:07 2019
# Command line used: -n -d /dev/lirc0
# Kernel version (uname -r): 4.9.41+
#
# Remote name (as of config file): RME
# Brand of remote device, the thing you hold in your hand:
# Remote device model nr:
# Remote device info url:
# Does remote device has a bundled capture device e. g., a
#     usb dongle? :
# For bundled USB devices: usb vendor id, product id
#     and device string (use dmesg or lsusb):
# Type of device controlled
#     (TV, VCR, Audio, DVD, Satellite, Cable, HTPC, ...) :
# Device(s) controlled by this remote:

begin remote

  name  RME
  bits           32
  flags SPACE_ENC|CONST_LENGTH
  eps            30
  aeps          100

  header       9031  4487
  one           565  1685
  zero          565   569
  ptrail        572
  repeat       9033  2252
  gap          107635
  toggle_bit_mask 0x0
  frequency    38000

      begin codes
          VolDown                  0x482C847B 0xBEAB0E1C
          VolUp                    0x482C44BB 0xBEAB0E1C
          Power                    0x482C04FB 0xBEAB0E1C
          Coax                     0x482CC43B 0xBEAB0E1C
          Opt                      0x482C24DB 0xBEAB0E1C
          USB                      0x482CA45B 0xBEAB0E1C
          Config1                  0x482C649B 0xBEAB0E1C
          Config2                  0x482CE41B 0xBEAB0E1C
          Config3                  0x482C14EB 0xBEAB0E1C
          Config4                  0x482C946B 0xBEAB0E1C
          Mute                     0x482C54AB 0xBEAB0E1C
      end codes

end remote

```

bengtmartensson, 2019-08-28:
NEC2-f16 is recognized very well, although it may be reported as something else according the the "prefer-over" parameter (see IrpProtocols.xml). 

Lirc, sigh... The signals generated does not correspond to a known protocol, possibly two different signals concatenated. Try IrpTransmogrifier from the command line instead. Applied to your VolDown command:
```
$ irptransmogrifier    decode --all  0000 006D 0044 0000 0157 00AB 0015 0016 0015 0040 0015 0016 0015 0016 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0016 0015 0040 0015 0040 0015 0016 0015 0016 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0016 0015 0016 0015 0016 0015 0040 0015 0040 0015 0040 0015 0040 0015 0016 0015 0040 0015 0040 0016 0658 0157 00AB 0015 0040 0015 0016 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0016 0015 0040 0015 0016 0015 0040 0015 0016 0015 0040 0015 0016 0015 0040 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0040 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0040 0015 0040 0015 0016 0015 0016 0016 05AF
JVC_alt: {D=18,F=52}, beg=0, end=35, reps=1 {UNDECODED. length=100}
  NEC: {D=18,F=33,S=52}, beg=0, end=67 {UNDECODED. length=68}
  NEC-Shirriff-32: {data=1210877051}, beg=0, end=67 {UNDECODED. length=68}
  NEC1-f16: {D=18,F=56865,S=52}, beg=0, end=67 {UNDECODED. length=68}
  NEC2: {D=18,F=33,S=52}, beg=0, end=67, reps=1 {UNDECODED. length=68}
  NEC2-f16: {D=18,F=56865,S=52}, beg=0, end=67, reps=1 {UNDECODED. length=68}
  Pioneer: {D=18,F=33,S=52}, beg=0, end=67, reps=1 {UNDECODED. length=68}
  Roku: {D=18,F=33,S=52}, beg=0, end=67 {UNDECODED. length=68}

irptransmogrifier    decode --all --recursive  0000 006D 0044 0000 0157 00AB 0015 0016 0015 0040 0015 0016 0015 0016 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0016 0015 0040 0015 0040 0015 0016 0015 0016 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0016 0015 0016 0015 0016 0015 0040 0015 0040 0015 0040 0015 0040 0015 0016 0015 0040 0015 0040 0016 0658 0157 00AB 0015 0040 0015 0016 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0016 0015 0040 0015 0016 0015 0040 0015 0016 0015 0040 0015 0016 0015 0040 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0040 0015 0040 0015 0016 0015 0016 0015 0016 0015 0016 0015 0040 0015 0040 0015 0040 0015 0016 0015 0016 0016 05AF
JVC_alt: {D=18,F=52}, beg=0, end=35, reps=1 {UNDECODED. length=100}
  NEC: {D=18,F=33,S=52}, beg=0, end=67 {NEC-Shirriff-32: {data=3198881308}, beg=68, end=135 NEC1-f16: {D=125,F=14448,S=213}, beg=68, end=135 NEC2-f16: {D=125,F=14448,S=213}, beg=68, end=135, reps=1}
  NEC-Shirriff-32: {data=1210877051}, beg=0, end=67 {NEC-Shirriff-32: {data=3198881308}, beg=68, end=135 NEC1-f16: {D=125,F=14448,S=213}, beg=68, end=135 NEC2-f16: {D=125,F=14448,S=213}, beg=68, end=135, reps=1}
  NEC1-f16: {D=18,F=56865,S=52}, beg=0, end=67 {NEC-Shirriff-32: {data=3198881308}, beg=68, end=135 NEC1-f16: {D=125,F=14448,S=213}, beg=68, end=135 NEC2-f16: {D=125,F=14448,S=213}, beg=68, end=135, reps=1}
  NEC2: {D=18,F=33,S=52}, beg=0, end=67, reps=1 {NEC-Shirriff-32: {data=3198881308}, beg=68, end=135 NEC1-f16: {D=125,F=14448,S=213}, beg=68, end=135 NEC2-f16: {D=125,F=14448,S=213}, beg=68, end=135, reps=1}
  NEC2-f16: {D=18,F=56865,S=52}, beg=0, end=67, reps=1 {NEC-Shirriff-32: {data=3198881308}, beg=68, end=135 NEC1-f16: {D=125,F=14448,S=213}, beg=68, end=135 NEC2-f16: {D=125,F=14448,S=213}, beg=68, end=135, reps=1}
  Pioneer: {D=18,F=33,S=52}, beg=0, end=67, reps=1 {NEC-Shirriff-32: {data=3198881308}, beg=68, end=135 NEC1-f16: {D=125,F=14448,S=213}, beg=68, end=135 NEC2-f16: {D=125,F=14448,S=213}, beg=68, end=135, reps=1}
  Roku: {D=18,F=33,S=52}, beg=0, end=67 {NEC-Shirriff-32: {data=3198881308}, beg=68, end=135 NEC1-f16: {D=125,F=14448,S=213}, beg=68, end=135 NEC2-f16: {D=125,F=14448,S=213}, beg=68, end=135, reps=1}
```
As you can see, IrScrutinizer does not give access to all the possibilities of IrpTransmogrifier.