bengtmartensson/IrScrutinizer #264: IrScrutinizer source build problem

Pavlex4, 2018-11-05:
I tried to IrScrutinizer project from source but I got below. Can you tell me how to fix this or send me all required jar files?

Failed to execute goal on project IrScrutinizer: Could not resolve dependencies for project org.harctoolbox:IrScrutinizer:jar:1.2: The following artifacts could not be resolved: org.harctoolbox:IrpMaster:jar:1.2, org.harctoolbox:HarcHardware:jar:1.2, org.harctoolbox:Girr:jar:1.2, org.harctoolbox:Jirc:jar:1.2, com.mrallen:tonto:jar:1.44, org.harctoolbox:Crystal-Clear-Icons:jar:20070620: Could not find artifact org.harctoolbox:IrpMaster:jar:1.2 in central (http://repo.maven.apache.org/maven2) -> [Help 1]

To see the full stack trace of the errors, re-run Maven with the -e switch.
Re-run Maven using the -X switch to enable full debug logging.

For more information about the errors and possible solutions, please read the following articles:
[Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException

bengtmartensson, 2018-11-06:
This is to, or at least should be, described in the manual, and [this except of the manual](https://github.com/bengtmartensson/harctoolboxbundle/blob/master/BUILD.md). Look at the scripts tools/build-* and .travis.yml. Note that to build 1.4.2 you need the branch `IrpMaster` of some of the dependencies.

Version 1.2 is stone-age,

bengtmartensson, 2018-11-06:
> Can you please send me tonto jar file?

That will have to wait until Friday, since I am on a business trip right now.

bengtmartensson, 2018-11-09:
As promised: [tonto.jar](http://www.harctoolbox.org/downloads/tonto.jar)

Pavlex4, 2018-11-09:
Thank you very much

Pavlex4, 2018-11-12:
Can you please send me built project because I am still having issues with building?

bengtmartensson, 2018-11-12:
Builds of latest snapshot as well as latest official release are always found [here](https://github.com/bengtmartensson/harctoolboxbundle/releases). 

For IrScuitinizer2 there are presently no builds.

Pavlex4, 2018-11-12:
I found there build but that is compiled jar. I need compiled source because I want to debug source code.

bengtmartensson, 2018-11-12:
I am afraid that I cannot make sense of the last sentence. Does it help you that a jar file is nothing but a zip archive of .class files?

Pavlex4, 2018-11-12:
I know what is jar. I just want built source so I can run it in netbeans and set breakpoints for debugging

bengtmartensson, 2018-11-13:
@Pavlex4 :
I am sorry, at this point I am not able to help you.

Pavlex4, 2018-11-13:
It doesn't matter. Anyway thanks again for help

closed 2018-11-16 by bengtmartensson