bengtmartensson/IrScrutinizer #267: How to analyse an unknown protocol of an AC remote?

magedrifaat, 2019-02-17:
I'm not sure if this is the right place to ask this but I have an old carrier AC remote that I have tried many libraries to analyse it but didn't get any helpful result. The closest I got is using IrScrutinizer as I was able to record and send the signals using an Arduino UNO.
But the problem is that every signal (or so I think, I am very new to this) sends the whole state of the remote not just the pressed button, So I can't have a "Temperature up" button or so (e.g. for an android app) without decoding the signal and analyzing which bit is which.
The file attached contains raw and pronto hex data from IrScrutinizer for a signal from the remote, if anymore data is needed just tell me how to get it and I will provide it.

[girr_2019-02-18_00-24-57.zip](https://github.com/bengtmartensson/harctoolboxbundle/files/2873490/girr_2019-02-18_00-24-57.zip)

bengtmartensson, 2019-02-18:
I am sorry, but there is not much I can do to help you. You already have succeeded in capturing a working signal, and it looks sane. It is definitely clean. It is a 72 bit payload modulated in the same way as for example a NEC protocol. So you are asking for the semantics of these 72 bytes... 

There are a number of projects on the Internet that tries to do this for different AC systems. Possibly someone has already done it for your system.

In case you want to invest time in it, capture all known signals. You might find [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier) useful for analysing the collection.

magedrifaat, 2019-02-18:
Hello,

Thanks for the quick reply. I actually found some very helpful resources and will start decoding it myself, then will possibly send the results somewhere to be supported in the future (is here suitable?).
What I need to find now is slightly off-topic and that is if there is actually an android application that I can use to implement the protocol once decoded or I will have to write one myself.

bengtmartensson, 2019-02-18:
Ideally, I would prefer to have the results in the form of parameterized IRP-protocols, see the protocol Fujitsu_Aircon in IrpTransmogrifier (IrpProtocols.xml) (although it still leaves things to be desired) Then, under some conditions, IrpTransmogrifier can generate code (C++/Java,... etc) for an implementation .

If you have the information, going down this path should not be too hard.

magedrifaat, 2019-02-18:
Is there a way to get the binary representation (the 72 bits) directly from IrScrutinizer?

bengtmartensson, 2019-02-18:
Sure, make sure that Options -> Invoke analyzer is selected. Then see the Analyze field in "Scrutinize signal" or in the Scrutinize remote -> Raw remote. 

You can also use IrpTransmogrifier..
```
$ irptransmogrifier analyze --maxparameter 72 0000 006D 004A 0000 0150 00AB 0015 0015 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0040 0015 0015 0015 0040 0015 0015 0015 0040 0015 0015 0015 0040 0015 0015 0015 0040 0015 0040 0015 0040 0015 0040 0015 0015 0015 0040 0015 0040 0015 0015 0015 0040 0015 0015 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0040 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0040 0015 0040 0015 0040 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0040 0015 0015 0015 0040 0015 0015 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0770
{38.0k,552,msb}<1,-1|1,-3>(16,-8,A:72,1,-50m){A=0x21aaf690a8ba061560}
```

magedrifaat, 2019-02-28:
Hello,
So I was able to analyze the protocol and after some painful time I understood the IRP notation and managed to formulate the protocol (albeit in a very ugly way). The excel file attached contains the full breakdown of the protocol in LSB. I also made an android application to control the AC from my phone but it is too ugly to be shared with anyone yet since it is my first attempt at android development so I might improve it a bit and then upload it to my account.
I would like to thank you very much for all your tools, they have made my life much easier in both analyzing the protocol and all the testing stages after that.
[Carrier protocol analysis.xlsx](https://github.com/bengtmartensson/harctoolboxbundle/files/2916853/Carrier.protocol.analysis.xlsx)

closed 2019-02-28 by magedrifaat