bengtmartensson/IrScrutinizer #276: Sharp RC1912

bakdamir, 2019-05-27:
Not determined.(+872 -872 +1749 -872 +872 -872 +872 -872 +872 -872 +872 -1749 +872 -61 +872 -872 +872 -1749 +872 -872 +1749 -872 +872 -24822).

bengtmartensson, 2019-05-27:
What are you trying to say?

The given signal is not supposed to produce a decode.

closed 2019-05-27 by bengtmartensson

bakdamir, 2019-05-27:
I'm not trying to say anything, but I assert that the program does not decode this remote control.

bengtmartensson, 2019-05-27:
Your signal is not to produce a decode, since it does not match any of the patters. Not to decode it is the correct behavior. 

The signal 
+872 -872 +1749 -872 +872 -872 +872 -872 +872 -872 +872 -1749 +872 -61 +872 -872 +872 -1749 +872 -872 +1749 -872 +872 -24822

is very unprobable and is almost surely a measurement error.  Where did you get it from? The "-61" makes no sense. If removing "-61" and replacing the adjacent numbers (+872) by their sum you get 
+872 -872 +1749 -872 +872 -872 +872 -872 +872 -872 +872 -1749 +1744 -872 +872 -1749 +872 -872 +1749 -872 +872 -24822

which decodes.

bakdamir, 2019-05-27:
This is a signal from the TV remote Sharp RC1912

bengtmartensson, 2019-05-27:
and how did you came up with the conclusion that it sends +872 -872 +1749 -872 +872 -872 +872 -872 +872 -872 +872 -1749 +872 -61 +872 -872 +872 -1749 +872 -872 +1749 -872 +872 -24822 ??

More explicitly: how did you measure? Can you reproduce the measurement?

bakdamir, 2019-05-27:
your program,IrScrutinizer

bengtmartensson, 2019-05-27:
What hardware did you use for the measurement?

bakdamir, 2019-05-27:
[image]

bakdamir, 2019-05-27:
excuse me
[image]

bakdamir, 2019-05-27:
the remote works according to the protocol rc5

bengtmartensson, 2019-05-28:
I have nothing more to add. Instead, let me summarize:

* First, if you request support or help, say so. Your initial posting definitely creates the impression that you claim that the program has a bug. (I am normally very helpful when people ask for my help, less so when people erroneously claim that my programs are buggy.)
* Your signal is simply senseless; not to produce a decode is the correct behavior. (Therefore "invalid".)
* As I wrote in the second post, the problem is almost surely the "-16". See also your own plot -- it is the "notch" at around 13000. Removing that manually, and it decodes to -- RC5, consistent with your last post.
* To possibly find the problem and help you, I asked for the hardware you are using, which you did not answer (possibly a language problem?)

bakdamir, 2019-05-29:
Yes, I asked for help. I constantly use your program, and I am very grateful to you for your titanic work. The equipment in the screenshot.
[image]

bakdamir, 2019-05-29:
ubuntu operating system 18.04,a laptop Acer Aspire 3,A315-51-32p6

bakdamir, 2019-05-29:
IrScrutinizer-1.4.2-x86_64.AppImage

bengtmartensson, 2019-05-29:
So, then why don't you just accept that the "-61" capture was a measurement error and redo it?

Please also let me recommend that you update GirsLite to  [this version](https://github.com/bengtmartensson/AGirs/releases/tag/Version-1.0.1).