bengtmartensson/IrScrutinizer #279: Bundle subset of JRE in the AppImage

probonopd, 2019-06-12:
We can now use https://adoptopenjdk.net/ to produce a subset of Java just enough to run IrScrutinizer.

The following illustrates how to do this for an application that is provided by upstream without a JRE.

```
wget -c https://github.com/AdoptOpenJDK/openjdk11-binaries/releases/download/jdk-11.0.3%2B7_openj9-0.14.3/OpenJDK11U-jdk_x64_linux_openj9_11.0.3_7_openj9-0.14.3.tar.gz
tar xf OpenJDK11U-jdk_x64_linux_openj9_11.0.3_7_openj9-0.14.3.tar.gz

wget -c https://github.com/bengtmartensson/harctoolboxbundle/releases/download/ci-build/IrScrutinizer-1.4.3-SNAPSHOT-x86_64.AppImage

sudo mount ./IrScrutinizer-1.4.3-SNAPSHOT-x86_64.AppImage /mnt
sudo cp -r /mnt/ AppDir && sudo chown -R $USER AppDir
sudo umount /mnt 

#
# Find out which subset of the Java classes need to be bundled
#

./jdk-11.0.3+7/bin/jdeps  --list-deps AppDir/usr/share/irscrutinizer/IrScrutinizer-jar-with-dependencies.jar

   java.base
   java.datatransfer
   java.desktop
   java.logging
   java.xml

# For other applications, it may be necessary to use --module-path, see
# https://medium.com/azulsystems/using-jlink-to-build-java-runtimes-for-non-modular-applications-9568c5e70ef4

#
# Create our custom subset of Java
#

./jdk-11.0.3+7/bin/jlink' --no-header-files --no-man-pages --compress=2 --strip-debug --add-modules java.base,java.datatransfer,java.desktop,java.logging,java.xml --output usr

#
# Copy it into the AppDir and re-create the AppImage
#

cp -Rf usr AppDir/

wget -c https://github.com/AppImage/AppImageKit/releases/download/12/appimagetool-x86_64.AppImage
chmod +x appimagetool-x86_64.AppImage 
./appimagetool-x86_64.AppImage AppDir/
```

The launcher script needs to be adjusted a bit, make sure to run `java -jar -Xquickstart` because otherwise Java optimizes for long-running processes, which means a slower start (roughly 5 rather than 3 seconds): https://www.eclipse.org/openj9/docs/xquickstart/

The resulting AppImage will be "just" 37 MB in size (compared to 76 MB without the subsetting).

bengtmartensson, 2019-06-13:
Thank you very much for this suggestion. What is needed is an automated procedure for building. It does not need to be integrated in the continuous integration build. Also licensing issues must be clarified.

This should be an alternative, not a replacement of the exising JDK-less AppImages.

PR anyone?

probonopd, 2019-06-13:
Would you like me to integrate this into the continuous build script? That'd be easier than creating a whole new different script different from the continuous builds.

bengtmartensson, 2019-06-13:
I am not quite sure what you mean by "the continuous build script". If you can extend or modify (my) tools/mkappimage.sh to generate an Appimage with embedded, minimal  JDK  alongside with the old one, I would be really interested.

bengtmartensson, 2019-06-15:
I see you checked in some stuff to a branch. Thanx a bunch!

I will first finish #280, then I will merge you fix. Thanx again.

probonopd, 2019-06-15:
Yes, `probonopd-patch-1`. I think the old `AppImage` branch can be deleted.

commit 4b7bc7be (2019-06-19): AppImage now comes (optionally) comes with bundled JDK.
Resolves #279.
closed 2019-06-19 by bengtmartensson