bengtmartensson/IrScrutinizer #284: IrScrutinizer GUI not loading on Raspbian GNU/Linux 9 Stretch

BStenfors, 2019-06-21:
For whoever can help me.

I have installed, I think, the binaries correctly for IrScrutinizer on my raspberry pi but when I load the app via command line the app loads but no GUI controls are visible.

There are no errors logged to the command line screen.

TIA,
S104S

bengtmartensson, 2019-06-21:
What is the output from 

    irscrutinizer --version

BStenfors, 2019-06-21:
IrScrutinizer version 1.4.3
JVM: Raspbian 1.8.0_212 Linux-arm

bengtmartensson, 2019-06-21:
Probably the Java is "funny". Can you verify that it works for graphical Java programs? Also try with another Java JVM, either Oracle or OpenJDK  (version >= 8).

BStenfors, 2019-06-21:
Awesome! Thanks for the tips.
Had to step away from my computer for a bit but will give those options a shot here shortly.

Thanks again!

Will let ya know how it goes!

BStenfors, 2019-06-23:
It appears that I cannot install a version of Java greater than 8 on a raspberry pi zero.  I have a raspberry pi 3 laying around I am going to see if I can get it going on.  Will keep you updated.

Thanks!
S104S

bengtmartensson, 2019-06-24:
Java 8 is OK, just my suspicion is you possily have a "headless" Java, or somehow a Java broken in some other way. Please try
    
    java -classpath /usr/local/share/irscrutinizer/IrScrutinizer-jar-with-dependencies.jar org.harctoolbox.guicomponents.StandalonePopupAnnoyer "hello world"

(possibly replacing /usr/local/share/irscrutinizer).

BStenfors, 2019-06-24:
Good news!!!

I think I must of had a "headless" version installed. 

I found an article to install a different version of 1.8 and everything loads!

Thanks for all your help!!

Cheers,
S104S

bengtmartensson, 2019-06-24:
That is great. Can I ask you to write a few lines (possibly with the link to the article) to help the next person who has the same problem?

Still I am slightly embarrassed from 

> when I load the app via command line the app loads but no GUI controls are visible.

there should be an error message on stderr at least...

bengtmartensson, 2019-06-24:
> there should be an error message on stderr at least...

When started "headlessly" the program gives a "friendly" error massage on stdout. 

    try {
          new GuiMain(applicationHome, propsfilename, verbose, userlevel, arguments).setVisible(true);
    } catch (HeadlessException ex) {
          System.err.println("This program does not run in headless mode.");
    }

When I just tried it, it worked as expected. Are you really sure you did not get that message?

BStenfors, 2019-06-24:
lol, don't be embarrassed!!

I don't recall seeing that message appearing but I also may have had a "bungled" installation of java as well as it being "headless".

I will definitely edit my post with the link to the article.

Here is the output of that command:

It gave me a error dialog with "hello world" with options of "Continue" or "Exit".

I pressed "Continue" and it just gave me a blank command prompt.

bengtmartensson, 2019-06-27:
> It gave me a error dialog with "hello world" with options of "Continue" or "Exit".
>
> I pressed "Continue" and it just gave me a blank command prompt.

This is/was just a trivial test of "headfullness".

bengtmartensson, 2019-07-02:
Closing as solved.

However, I still do not know whether the headless detection can or needs to be improved.

closed 2019-07-02 by bengtmartensson