bengtmartensson/IrScrutinizer #285: Support command  line program IrpTransmogrifier in AppImage

bengtmartensson, 2019-06-27:
The "package" comes with the GUI program IrScrutinizer and the command line program [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier). Both are installed (made available as commands) with Windows and the generic binary distribution. Leaves AppImage and the Mac app, which presently only offers the GUI. I do not care right now for the Mac...

@probonopd , comments?

probonopd, 2019-06-27:
You can put it into the AppImage and then use something along those lines:

https://gitlab.com/inkscape/inkscape/blob/master/packaging/appimage/AppRun#L8-29

closed 2019-06-28 by bengtmartensson

bengtmartensson, 2019-06-28:
Should work now.

(Preliminary) documentation:

If the AppImage is invoked with basename `irptransmogrifier` (note: all lowercase), instead the command line program IrpTransmogrifier will be invoked on the arguments. Alternatively, if the AppImage is invoked with the first argument (exactly) equal to `irptransmogrifier`, that program will be invoked on the remaining arguments.

@probonopd : Thanx for your support!

commit 6120eeb3 (2019-06-28): Support calling IrpTransmogrifier in AppImage. Resolves #285.