bengtmartensson/IrScrutinizer #290: IrScrutinizer-2.0.0-8bf4199-SNAPSHOT.exe capture and export errors

DanielTheGreat, 2019-07-13:
**Issue 1:**
Start IrScrutinizer
Open GirsLite device
Scrutinize remote tab
Parametric Remote
Add capture name (e.g. 'Test')
Capture
Fire some IR commands (from JVC RM-SRX8032R) at the Nano device's sensor
Error (in console pane) on each capture: "Undecodable signal, ignored" (these same signals were decoded by v1.4.3)
Capture (off)

**Issue 2:** (continuing from above settings, changed as below)
Raw Remote
Capture
Fire some IR commands (from JVC RM-SRX8032R) at the Nano device's sensor
Capture (off)
Enter command names in Name column
Export tab
Export dir. (default): C:\Users\Daniel\AppData\Local\Temp\\exports (note double-backslash!)
Format: Arduino/Infrared4Arduino
Girr export tab
Stylesheet (default): simplehtml.xsl
Export raw remote
Error pop-up: "C:\Users\Daniel\AppData\Local\Temp\exports\arduino\infrared4arduino_2019-07-13_16-44-49.ini (The system cannot find the path..."
[image]

bengtmartensson, 2019-07-13:
Please open one issue per ... issue.

1. How do you envision that I should analyze this? Please post the non-decoding signals, preferably in a new issue.

2. Thank you very much for this. Turned out that it was dependent on the "/" in the format name; the problem occurred only on Windows. Should work in the most current snapshot. Please verify.

bengtmartensson, 2019-07-13:
You have made a separate issue for issue 1, and there report the export as working. So I am closing this as fixed.

closed 2019-07-13 by bengtmartensson