bengtmartensson/IrScrutinizer #291: Overlapping text in signal graphic

probonopd, 2019-07-13:
Entering and scrutinizing

`+8000, -8000, +2000, -1000, +1000, -1000, +2000, -1000, +2000, -1000, +1000, -1000, +1000, -1000, +1000, -1000, +1000, -1000, +2000, -1000, +2000, -1000, +1000, -1000, +2000, -1000, +1000, -2000`

I get

[image]

The last number is not displayed clearly because of overlapping text.

bengtmartensson, 2019-07-13:
Thank you for the report. Although it is hard to argue that the behavior is correct, it is not a KO-issue. There are a few issues, an missing features with the plotter, this is not the most urgent :-).

probonopd, 2019-07-13:
Indeed :)

commit ed76718f (2024-01-05): Fix glitch in plot scales, leading to overlapping text. Resolves #291.
closed 2024-01-05 by bengtmartensson