bengtmartensson/IrScrutinizer #292: IrScrutinizer-2.0.0-0f02bf2-SNAPSHOT.exe capture errors

DanielTheGreat, 2019-07-13:
"Undecodable signal, ignored" when trying to capture parametric remote data, as reported in Issue #290.
'Arduino/infrared4Arduino' export now working, so attached raw export for signals not recognised in parametric capture.
Would have sent as Girr export, but that's not working in this version.  (Sorry, no time to lodge another Issue for that; let's just fix one thing at a time and I'll lodge another issue after that.)
[arduino_infrared4arduino_2019-07-13_23-32-40_ino.txt](https://github.com/bengtmartensson/harctoolboxbundle/files/3388832/arduino_infrared4arduino_2019-07-13_23-32-40_ino.txt)

bengtmartensson, 2019-07-13:
Daniel, if anyone asks you  for a raw signal, just cut-n-paste it, or do a text export as pronto or raw (not forgetting the frequency).

Having said that, I randomly checked the first and the last, and they decode ... (to JVC{2]).

> Would have sent as Girr export, but that's not working in this version.

Looking forward to your bug report.

bengtmartensson, 2019-07-16:
@DanielTheGreat : please try the current snapshot, and let me know if that fixes the problem.

DanielTheGreat, 2019-07-17:
Bengt,

Still not decoding my JVC remote (see attached screen-shot).

Regards,
Daniel

bengtmartensson, 2019-07-17:
I can find no screenshot. Please cut and paste the offending signal as Pronto Hex

DanielTheGreat, 2019-07-17:
Bengt,

1	17:35:18	+8419,-4235	+520,-1615,+520,-1615,+520,-520,+520,-520,+520,-520,+520,-1615,+520,-520,+520,-1615,+520,-520,+520,-1615,+520,-1615,+520,-1615,+520,-1615,+520,-520,+520,-520,+520,-520,+520,-20448					false		38461.0
2	17:35:20	+8419,-4236	+518,-1615,+518,-1615,+518,-518,+518,-518,+518,-518,+518,-1615,+518,-518,+518,-1615,+518,-1615,+518,-1615,+518,-1615,+518,-1615,+518,-1615,+518,-518,+518,-518,+518,-518,+518,-19392					false		38461.0
3	17:35:21	+8419,-4235	+523,-1615,+523,-1615,+523,-523,+523,-523,+523,-523,+523,-523,+523,-523,+523,-523,+523,-523,+523,-1615,+523,-1615,+523,-1615,+523,-1615,+523,-523,+523,-523,+523,-523,+523,-22576					false		38461.0
4	17:35:22	+8419,-4235	+521,-1615,+521,-1615,+521,-521,+521,-521,+521,-521,+521,-521,+521,-521,+521,-521,+521,-1615,+521,-1615,+521,-1615,+521,-1615,+521,-1615,+521,-521,+521,-521,+521,-521,+521,-21520					false		38461.0

Regards,
Daniel

bengtmartensson, 2019-07-17:
What happens is probably the following: The repeat finder reduces the JVC protocols so that the decoder does not recognize it. This is due to the particular nature of that protocol.You can try without the remote finder. (Options -> Invoke repeat finder).

The current snapshot recognizes it however as "JVC_alt". Please try it.

DanielTheGreat, 2019-07-18:
Bengt,

On the previous Snapshot (IrScrutinizer-2.0.0-38e5c14-SNAPSHOT), it did not capture as Parametric Remote whether 'Invoke repeat finder' was on or off ('Undecodable signal, ignored').

With the latest snapshot (IrScrutinizer-2.0.0-fda1706-SNAPSHOT) it does capture the Parametric with 'Invoke repeat finder' enabled, but is a bit 'flaky'...

After successfully capturing (Parametric) a few buttons, I stop the Capture, disable 'Invoke repeat finder', start Capture again.  'Undecodable signal, ignored'.  Stop Capture, re-enable 'Invoke repeat finder', try to start Capture a third time but get the attached 'Another capture thread is running' error pop-up.  After some time (maybe > 30-sec?) try Capture button again, and it's OK.  But still 'Undecodable signal, ignored'.

Close IrScrutinizer, ignoring the 'unsaved' message, and then start it again.  But despite 'Invoke repeat finder' still being enabled, I now get 'Undecodable signal, ignored' forever after, no matter how 'Invoke repeat finder' is set.  To recover, I close IrScrutinizer, unplug the Nano for several seconds, plug it in again and start IrScrutinizer again.  Then we're back to normal behaviour.  Well, for the first few tries, anyway. :-(

Now I shut down IrScrutinizer, power-cycle the Nano, re-start IrScrutinizer (with 'Invoke repeat finder' still enabled as default), but now always get 'Undecodable signal, ignored'.  I'm giving up at this stage; other things need to be done.

Regards,
Daniel

bengtmartensson, 2019-07-18:
The way to investigate a decoding problem is to use ONE recorded signal and send it to the decoder, for example using the "Scrutinize signal" pane. That way "experiments" are made reproducible (opposite of "flaky"). Saying that "this signal ... ...  does not decode the expected way" can be acted upon, your description above can not.

DanielTheGreat, 2019-07-18:
Bengt,

With a fresher mind in the morning, I'll see if I can work out what you're asking for, and make another screen-video for you.

Regards,
Daniel

bengtmartensson, 2019-07-18:
1. turn on Options -> Output text format -> Pronto hex
2. turn off Options -> repeat finder
3. turn on Options -> "Print decodes to console"
4. configure and open hardware
5. Go to Scrutinize signal
6. Press "Capture" and shoot one IR signal. (ONE!)
7. Cut-n-paste data window and console window to post. (I don_t want to OCR your screendumps!) 

If desired or necessary, repeat 6-7 for other signals, try with and without repeat finder, possibly also with/without signal cleaner.

DanielTheGreat, 2019-07-19:
Bengt,

As per your clear instructions, data window (I presume) for Vol+:

0000 006C 0056 0000 0144 00A3 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 003E 0013 0015 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 02EA 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 003E 0013 0015 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 02EA 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 003E 0013 0015 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 02EA 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 003E 0013 0015 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 02EA 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 003E 0013 0015 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 09D9

Console window (I presume):

JVC: {D=163,F=31}, beg=0, end=35

-----

Then with 'Invoke repeat finder' enabled, for Vol-:

0000 006C 0001 0011 0144 00A3 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 003E 0013 0015 0013 003E 0013 0015 0013 003E 0013 003E 0013 003E 0013 003E 0013 0015 0013 0015 0013 0015 0013 0312

JVC_alt: {D=163,F=30}

Let me know if you want more.

Regards,
Daniel

bengtmartensson, 2019-07-19:
Looks perfect to me. Wouldn't you agree to consider the problem solved and close the issue?

DanielTheGreat, 2019-07-19:
Bengt,

Strangely, it does seem to be working better today.  So go ahead and close this issue, thanks.

Regards,
Daniel

bengtmartensson, 2019-07-19:
OK, I consider it solved. To help people browsing the items, I have changed the title.

closed 2019-07-19 by bengtmartensson