bengtmartensson/IrScrutinizer #293: IrScrutinizer-2.0.0-0f02bf2-SNAPSHOT.exe Girr export error

DanielTheGreat, 2019-07-14:
When exporting a Girr file, a zero-byte file is exported, and not fully written till IrScrutinizer closed.
On Win7Pro64.

[IrScrutinizer_Girr_export_bug_20190714_13h58m11s.zip](https://github.com/bengtmartensson/harctoolboxbundle/files/3389529/IrScrutinizer_Girr_export_bug_20190714_13h58m11s.zip)

bengtmartensson, 2019-07-14:
When I try to reproduce on Win10, 32bit, German, the following happens:

A file is created, "0 KB". However, when I open it with a text editor, its content is there, correctly. Only when I close IrScrutinizer, the size will be > 0. Please verify. You can also check if the "Text" export format has the same behavior. 

Have you discovered what F9 does in the remote tables? Very handy when trying exporting etc...

Manufacturer etc is just "decoration", if the receiver of the file (e.g. a collection of such data) requests such information. You can just turn it off by unselecting Options -> Export options -> Inquire metadata.

DanielTheGreat, 2019-07-14:
Yes, same here.  EXCEPT that after opening the .girr with a text editor, then closing the editor, a refresh of the Xplorer2 window will now show the correct filesize, even though I haven't closed IrSrutinizer.
Text export appears to work correctly, immediately showing the filesize without doing anything first.
I remember reading about F9 for filling-in arbitrary data, but hadn't tried it.  Yes, useful for quick export tests!  Thanks for the reminder.

bengtmartensson, 2019-07-14:
So it does not appear to be an IrScrutinitzer bug after all, possibly a Windows  bug (optimization taken too far). OK to close?

DanielTheGreat, 2019-07-14:
I guess so.  Thanks.

closed 2019-07-14 by DanielTheGreat