bengtmartensson/IrScrutinizer #304: RC6A

Cyrillevergely, 2019-08-02:
I need to generate some RC6A codes for a KNX device
Original remote TK 124 made by Philips use RC6A
The protocol is RC6A / Device 7 / Subdevice 118 / OBC 1 to 24

Unfortunately i can't find RC6A in the list of the protocol in the IR Scrutinizer 
Some codes are recognized as Protocol "Replay" by IR Scrutinizer but they are not working.

Here are some information found about RC6A :
_In the RC6A code the system code is 8 bits allowing for values of 0-255.
It's very hard to find piece of information about RC6A. RC6A is the code used in the remotes for the Microsoft MCE system

The RC6A format itself is defined as follows (represented as a bit stream):
hhhhhhhh TT s UUUUUUU SSSSSSSS CCCCCCCC or
hhhhhhhh TT s UUUUUUUUUUUUUUU SSSSSSSS CCCCCCCC

hhhhhhhh Header data (17) (NOT to be biphase encoded), ‘111111100 10 10 10 01' 9 bits of AGC control, Consists of 7 on times and 2 off times or 3.1115 msec on and 889 usec off.

one stop bit always (1) 444.5usec on, 444.5 usec off , 3 bits giving code number (6 in this case as 10 10 01) ergo 10= 444.5 usec on 444.5 usec off

TT Toggle bits (4) (NOT to be biphase encoded)
‘0011' or 889usec off, 889usec on
‘1100' 889usec on, 889usec off

s = 0 Customer range 0..127 (to be biphase encoded)
s = 1 Customer range 32768..65535 (to be biphase encoded)
UUUUUUU CUSTOMER identifier (7 or 15) (to be biphase encoded)
7 or 15 bits depending on the ‘s’ bit
SSSSSSSS System bits (8) (to be biphase encoded)
CCCCCCCC Command bits (8) (to be biphase encoded)
All bits to be biphase encoded will have a single bit split up into two half bits:
0 -> 01
1 -> 10 Note: inverse from RC5/RC5X biphase format!
Each half bit time is 444.5 microseconds__

bengtmartensson, 2019-08-02:
Can you capture a few of those signals and post it here?

Cyrillevergely, 2019-08-02:
Hello

Yes of course
attached the spreadsheet
It's recognized as replay by the best software ever the famous IR Scrutinizer
I love this software

By the way i work for RTI :-)

Cyrille VERGELY
European Account Executive
Dealer Experience Europe
Remote Technologies Incorporated
Cell: +33 6 48 128 128

[cid:3bfc24f3-0afd-4024-8e9a-b05971f73af7@namprd11.prod.outlook.com]

[cid:a5e9b501-e233-47ac-85bd-e9ed3ed6fa5e@namprd11.prod.outlook.com]

bengtmartensson, 2019-08-02:
Sorry, there is neither an attachment nor a link..

Cyrillevergely, 2019-08-02:
I have attached a csv file on the first email ...

Here is again as a link <https://drive.google.com/open?id=1jiDst_39ir0CJvMKVAwxHc6U5rSfeTok> also
 
Cyrille VERGELY
Mail: vergely.cyrille@gmail.com <mailto:vergely.cyrille@gmail.com>
Mobile : +33 6 48 12 81 28

bengtmartensson, 2019-08-02:
All those file decodes as "Replay". The comment for that protocol also says:

> {36k,444,msb}<-1,1|1,-1>(6,-2,1:1,6:3,<-2,2|2,-2>(T:1),D:8,S:8,F:8,-100m/*???*/)+[D:0..255,S:0..255,F:0..255,T@:0..1=0]

> Replay is a member of the RC6 family.  Technically it is RC6-6-24, but DecodeIr will always display the name as "Replay".  ProntoEdit calls this protocol "RC6 mode 6A" and KM has it under the alternate name "RC-6a" as well as its primary name "Replay".  RM has it under the alternate name "RC6-M-24n" as well as its primary name "Replay".

> In ProntoEdit, DecodeIr's "Device" is called "Customer Code"; DecodeIr's "Subdevice" is called "System"; and DecodeIr's "OBC" is called "Command".

So that settles the question "how to generate RC6A?" If 

> but they are not working

it must have another reason, for example that the receiver expects a toggle to be toggling, but the sent ones do not toggle.

Cyrillevergely, 2019-08-02:
Hello

Thank you so much for you replies
We used a toggle commande sending Toggle =0 and Toggle = 1

Only some where working the pair ones like 2 4 6 8 10 12
Impossible to get the function 7 working for exemple using the same method that generate the function 12 that worked.

Also check that the subdivice is not the same when Pair and Unpair...
That's strange.
All Pair commands works the unpair do not...

We will investigate...

Thanks for your prompt reply

Kind regards

Cyrille VERGELY
Mail: vergely.cyrille@gmail.com <mailto:vergely.cyrille@gmail.com>
Mobile : +33 6 48 12 81 28

bengtmartensson, 2019-08-03:
I added "RC6A" as synonym for Replay in the commit mentioned above.

Since the original question has been settled, I think it is appropriate to close this item. Feel free to open a new one if needed.

It ls possible that your device and/or remote has a different "understanding" than our Replay protocol. Against that theory speaks that all of your signals decode perfectly to (our) Reply/RC6A.

Cyrillevergely, 2019-08-05:
Hello Bengt

Thank-you so much for your prompt response.
I am happy that my post added some thoughts about RC6A
I was not aware about RC6A before :-)

The funny thing is that all pair commands works once generated by the IR Scrutinizer using Replay protocol
All are working very fine using Toggle Bit of course.

All impair commands are not working...
I noticed that in the learning of the original remotes all impair commandes do have a different Subdevice Number.
Very strange (attached is the spreadsheet of commands)

On the Job-site the dealer have used all pair commands and that's done.

Kind regards and many thanks

Cyrille VERGELY
Mail: vergely.cyrille@gmail.com <mailto:vergely.cyrille@gmail.com>
Mobile : +33 6 48 12 81 28

bengtmartensson, 2019-08-05:
> (attached is the spreadsheet of commands)

Again, there is no attachment. Please learn how to upload attachments to Github.
And please use Girr-exports from IrScrutinizer, NOT spreadsheets.

> On the Job-site the dealer have used all pair commands and that's done.

Am I supposed to understand?
* What job-site?
* What dealer?
* What is a pair/impair command?

closed 2019-08-08 by bengtmartensson