bengtmartensson/IrScrutinizer #312: Render broadlink_base64

k0rtina, 2019-08-15:
Are you able to add the ability to render to broadlink_base64?

This project already does it, but really only does NEC codes (as a source protocol): https://github.com/elupus/irgen

It would be nice to be able to do something like:
irptransmogrifier render --broadlink_base64 -n "D=33, S=160, F=142" Nokia32

At the moment I can use IrScrutinizer to turn:
protocol = Nokia32, device = 33, subdevice = 160, obc = 142, misc = X=38 T=0		false		36000

into the RAW:
+417 -278 +167 -278 +167 -611 +167 -278 +167 -444 +167 -611 +167 -611 +167 -278 +167 -278 +167 -278 +167 -611 +167 -444 +167 -611 +167 -611 +167 -278 +167 -778 +167 -611 +167 -88861

Then use irgen to convert the RAW to Broadlink Base64
irgen -i raw -d +417 -278 +167 -278 +167 -611 +167 -278 +167 -444 +167 -611 +167 -611 +167 -278 +167 -278 +167 -278 +167 -611 +167 -444 +167 -611 +167 -611 +167 -278 +167 -778 +167 -611 +167 -88861 -o broadlink_base64

Result:
JgBGAJSUEjcSNxI3EhISEhISEhISEhI3EjcSNxISEhISEhISEhISEhI3EjcSNxI3EjcSEhI3EjcSEhISEhISEhISEjcSEhIABfcNBQAAAAAAAAAAAAAAAAAAAAA=

This can then be used in something like HomeAssistant.

bengtmartensson, 2019-08-18:
Please evaluate the enclosed export format. Just unzip and drop into the exportformats.d directory.

[Homeass_broadlink.xml.zip](https://github.com/bengtmartensson/harctoolboxbundle/files/3513073/Homeass_broadlink.xml.zip)

bengtmartensson, 2019-08-21:
@k0rtina : politely requesting feedback on the solution to your problem that I presented above.

k0rtina, 2019-08-21:
I have completed some basic testing with mixed results. The generated codes are different to irgen, different to what I captured... but still seem to work!? (not sure why)

Example:
Yamaha Amp Power ON (nec1_D122F29)
Homeass_broadlink.xml (works) = JgBIAAABKJQSEhI3EhISNxI3EjcSNxISEjcSEhI3EhISEhISEhISNxI3EhISNxI3EjcSEhISEhISEhI3EhISEhISEjcSNxI3EgAFFwABKEoSAA0F

Captured IR (works)= JgBYAAABJ5USExI3EjgTNxI4ETkRORITETkSEhITEhMSExITERQRORITETkSNxM3EzcSOBI4EhMROREUERQSEhITEhMSExI4EgAFIgABKEoSAAxXAAEnSxIADQU=

IrScrutinizer (raw), converted with irgen (works) = 
JgBQAAABKJQSEhI3EhISNxI3EjcSNxISEjcSEhI3EhISEhISEhISNxI3EhISNxI3EjcSEhISEhISEhI3EhISEhISEjcSNxI3EgAFGQABKEoSAAxVDQUAAAAAAAA=

Yamaha Amp Power OFF (nec1_D122F30)
Homeass_broadlink.xml (works) = 
JgBIAAABKJQSEhI3EhISNxI3EjcSNxISEjcSEhI3EhISEhISEhISNxISEjcSNxI3EjcSEhISEhISNxISEhISEhISEjcSNxI3EgAFFwABKEoSAA0F

Captured IR (works)= 
JgBYAAABJ5QSExE5EjgRORE4EzcSOBITEjgRFBEUERQRFBEUERMSOBI4EjgRORE5ETkRORE4EhMSExITEhMRFBEUERQRFBE5EQAFIgABJ0sSAAxXAAEnSxIADQU=

IrScrutinizer (raw), converted with irgen (works) = 
JgBQAAABKJQSEhI3EhISNxI3EjcSNxISEjcSEhI3EhISEhISEhISNxISEjcSNxI3EjcSEhISEhISNxISEhISEhISEjcSNxI3EgAFGQABKEoSAAxVDQUAAAAAAAA=

*note* the power ON command is slightly different where it seems to trigger 'zone2' on the amp... so the captured signal might be a different button code.

bengtmartensson, 2019-08-21:
You are comparing apples to oranges: the captured signals contain measurement errors ("are dirty") (in particular, the Broadlink appears not to be a very good capturer), and in general a different number of repeats. 

I am reasonably confident about the numerics of the computing (but who knows?). But I am most interested in your feedback about the format of the generated file; should be something that Homeassistant can digest in one single chunk.

k0rtina, 2019-08-21:
The format is very usable, and is great to copy/paste from. The inclusion of the identifying code is great to help remind the user of what the code is for.

Here is the config from switchs.yaml for my rumpus room, it will show you the yaml format for the entire Broadlink.

I am happy with the format, but you could turn it into a few more lines with the entire broadlink device

- platform: broadlink #Rumpus
  host: 192.168.1.100
  mac: "78:0f:77:00:xx:xx"
  timeout: 15
  type: rm_mini
  switches:
    rumpus_panasonic_tv:
      friendly_name: "Rumpus TV Power"
      command_on: 'JgAyAXI5Dw4OKw4PDQ8ODw0PDg8ODw0PDg8NDw4PDQ8OLA0PDg8NDw4PDQ8ODw0PDg8ODw0sDg4ODw0PDg8ODw0PDg8NDw4PDSwOKw8qDyoPKw4ODg8NDw4sDisOKw4rDisPDg0sDgAJfnE6Dg8NLA0PDg8ODw0PDg8NDw4PDQ8ODw0PDg8OKw8ODQ8ODw0PDg8NEA0PDg8NDw4rDg8NDw4PDg8NDw4PDQ8ODw0PDiwNLA4rDisOKw8ODQ8ODw0sDyoPKg8rDisODg4sDgAJfXI6Dg4OLA4ODg8NDw4PDQ8ODw0PDg8ODw0PDg8NLA4ODg8NDw4PDg8NDw4PDQ8ODw0sDg8NDw4PDQ8ODw0PDg8NDw4PDisPKg4rDysOKw4ODg8ODw0sDisOKw8qDyoODw4rDgANBQAAAAAAAA=='
      command_off: 'JgCYAXE6Dg8NLA4ODg8NDw4PDQ8ODw4PDQ8ODw0PDg8NLA4PDQ8ODw0PDg8NDw4PDRANDw4rDg8ODg8ODg4ODw4PDQ8ODw0sDisOKw8qDiwOKw4ODg8NLA4rDyoPKw4rDisODg4sDgAJfXI6Dg4OLA0PDg8NDw4PDg4ODw4PDQ8ODw0PDg8NLA4ODg8ODw0PDw4NDw4PDQ8ODw4rDw4NDw4PDQ8ODw4PDQ8ODw0sDisOKw8qDyoPKw4ODg8NLA4rDyoPKw4rDisODg4sDQAJfnM5Dg4OLA0PDw4NDw4PDg4ODw4PDQ8ODw0PDg8NLA0PDg8ODw0PDg8NDw4PDQ8ODw4rDg8NDw4PDQ8ODw0PDg8ODw0sDSwOKw4rDyoOLA4ODg8NLA4rDisPKg8rDisODg4sDQAJfnI6Dg4OLA4ODg8NDw4PDQ8ODw4PDQ8ODw0PDg8NLA4PDQ8ODw0PDg8NDw4PDQ8ODw4rDg8NDw4PDQ8ODw4ODg8ODw0sDisOKw8qDisPKw4ODg8NLA4rDyoPKw4rDisODg4sDQANBQ=='
    rumpus_panasonic_tv_xbox_input:  #HDMI1
      friendly_name: "TV Xbox Input"
      command_on: 'JgDMAHI6DQ8OKw8ODRANDw8PDQ4ODw0QDg4NEA4ODRANLA0PDhAMDw8ODhAMEA4PDQ4ODw0sDg8NDw4rDw8MDw4QDBAODg4PDQ8PDg0PDisOLA0QDSsPDwwQDisNEA0rDiwODg4QDAAJfnI6Dg8OKw0PDhAMEA4ODg4ODw4PDg4ODw0PDhAMLA4ODw4ODw0QDg4NEA4ODQ8ODw0sDg8ODw0rDg8NEQ0ODhANDw0PDRANDw4ODysNLA0QDSwNDw4PDSwNDw4sDSwNDw8PDAANBQAAAAAAAAAAAAAAAA=='
    rumpus_panasonic_tv_kodi_input:  #HDMI2
      friendly_name: "TV Kodi Input"
      command_on: 'JgAyAXE6Dg8NLA0PDg8ODg8PDQ8NDw4QDA8ODw4ODw8NKw4PDQ8ODw0QDQ8NEA4ODhANDg4rDw8MEA0sDQ8ODw4ODg8NEA0sDg4PDg0QDSwNLA0PDisOLA0PDisOEA0rDisPDg4PDQAJfnI6Dg4OKw8ODQ8ODw4PDRAMEQ0PDQ8NDw4PDRANLA4PDQ8NDw4PDRANDw0QDRAODwwsDg4PDg0sDg8NEA4ODRANEAwsDw4NDw8ODSwOKw4QDCwNLA4PDisODw0sDSwODhANDQAJf3E6Dg8NLA4ODg8NDw8PDQ4PDw4ODg4ODw0PDhANKw4PDg4OEAwQDQ8NDw8ODRAODw0rDg8NDw8qEA4NEA0PDRANDg8rDg4ODw0PDisOKw8ODisOKw4QDSsPDwwsDSwOEAwPDgANBQAAAAAAAA=='      
    rumpus_panasonic_tv_tv_input:  #TV
      command_on: 'JgDMAHI6Dw4PKw0QDg4PDg4PDg8NEA0PDg8ODw4PDRANLA4PDg8ODw0PDg8PDg4PDRANDw4sDg8NLA4PDg8NEA0PDg8ODw8ODRANDw4PDiwOKw8ODg8ODw0sDg8PDg4rDysODw4rDgAJlXE6Dg8OLA0QDQ8ODw4PDg8NEA0PDg8ODw4PDRANLA4PDg8NEA0PEA0ODw8ODQ8PDg4sDg8NLA4PDw4ODw4ODg8ODw4PDg8NDw4PDiwOKw8ODg8ODw0sDg8ODw4sDSwODw4sDQANBQAAAAAAAAAAAAAAAA=='  
    rumpus_yamaha_amp_vol_up:
      command_on: 'JgBQAAABJ5QTEhI4EhMSOBE5ETkROREUETgSExI4EhMSExEUERQROREUETkRExI4EjgSExITERQROREUETkRExITEjgSOBI4EQAFIgABJ0wRAA0FAAAAAAAAAAA='
    rumpus_yamaha_amp_vol_down:
      command_on: 'JgBQAAABJ5QSExE5ERQRORE5ETgSOBITEjgRFBE5ERQRFBEUERMSOBI4EjgSExE5ETkRFBEUERMSExITEjgSExEUETkRORE5EQAFIgABJ0sSAA0FAAAAAAAAAAA='
    rumpus_yamaha_amp_vol_mute:
      command_on: 'JgBQAAABJ5QSExE5ERQSOBI4ETgSOBITEjgSExE5ERQRFBEUERMSOBITEhMSOBE5EjgRFBEUERQROBI4EhMSExEUETkRORI4EQAFIgABJ0sSAA0FAAAAAAAAAAA='
    rumpus_yamaha_amp:
      friendly_name: "Rumpus Amp Power"
      command_on: 'JgBYAAABJ5USExI3EjgTNxI4ETkRORITETkSEhITEhMSExITERQRORITETkSNxM3EzcSOBI4EhMROREUERQSEhITEhMSExI4EgAFIgABKEoSAAxXAAEnSxIADQU='
      command_off: 'JgBYAAABJ5QSExE5EjgRORE4EzcSOBITEjgRFBEUERQRFBEUERMSOBI4EjgRORE5ETkRORE4EhMSExITEhMRFBEUERQRFBE5EQAFIgABJ0sSAAxXAAEnSxIADQU='
    rumpus_yamaha_amp_kodi_input:
      friendly_name: "Amp Kodi Input"
      command_on: 'JgBYAAABJ5QTEhI4ERQRORE5ETkROBITEjgSExI4ERQRFBEUERQROBITEhMSOBITEjgRFBE5ERQROBI4EhMSOBITEjgRFBE5EQAFIgABJ0sSAAxXAAEnSxIADQU='
    rumpus_yamaha_amp_castaudio_input:
      friendly_name: "Amp Cast Audio Input"
      command_on: 'JgBYAAABJ5QSExI4ERQRORE5ETgSOBITEjgSExE5ERQRFBEUERQROBI4EhMSExI4ERQRFBE5ETkRExI4EjgSExI4ETkRFBEUEQAFIgABJ0sSAAxXAAEnSxIADQU='

k0rtina, 2019-08-21:
the template could produce something like this:
`- platform: broadlink 		###RoomName = Rumpus
  host: 1.2.3.4			##IP Address of the Broadlink
  mac: "78:0f:77:00:xx:xx"	##MAC Address of the Broadlink
  timeout: 15
  type: rm_mini
  switches:
    nec1_D122F29:
      friendly_name: "nec1_D122F29"	#Update the friendly name with what the code does (EG: "Rumpus Amp Power")
      command_on: 'JgBIAAABKJQSEhI3EhISNxI3EjcSNxISEjcSEhI3EhISEhISEhISNxI3EhISNxI3EjcSEhISEhISEhI3EhISEhISEjcSNxI3EgAFFwABKEoSAA0F'
      command_off: ''`

bengtmartensson, 2019-08-23:
Hmm, not my idea of a good design... Everything is a switch...

How about this: 
[Homeass_broadlink.xml.zip](https://github.com/bengtmartensson/harctoolboxbundle/files/3533886/Homeass_broadlink.xml.zip)

k0rtina, 2019-08-25:
sorry for the delayed reply (travelling).

That looks great. Should make it very easy for Home Assistant users.

bengtmartensson, 2019-08-26:
OK. Thanx for helping improving IrScrutinizer!!

closed 2019-08-26 by bengtmartensson

Montreal666, 2023-08-08:
The Home assistant Broadlink switch yaml format seems to have changed since the IRScrutinizer export xml was initially released:

[https://www.home-assistant.io/integrations/broadlink#switch](url)

`# Example configuration.yaml entry
switch:
  - platform: broadlink
    mac: MAC_ADDRESS
    switches:
      - name: Philips TV
        command_on: JgAcAB0dHB44HhweGx4cHR06HB0cHhwdHB8bHhwADQUAAAAAAAAAAAAAAAA=
        command_off: JgAaABweOR4bHhwdHB4dHRw6HhsdHR0dOTocAA0FAAAAAAAAAAAAAAAAAAA=
      - name: LG TV
        command_on: JgBYAAABIJISExETETcSEhISEhQQFBETETcROBESEjcRNhM1EjcTNRMTERISNxEUERMSExE2EjYSNhM2EhIROBE3ETcREhITEgAFGwABH0oSAAwzAAEfShEADQU=
        command_off: JgBYAAABIJISExETETcSEhISEhQQFBETETcROBESEjcRNhM1EjcTNRMTERISNxEUERMSExE2EjYSNhM2EhIROBE3ETcREhITEgAFGwABH0oSAAwzAAEfShEADQU=`

Could the xml be updated to reflect this new format?

Thanks!

Montreal666, 2023-08-08:
Also, it would be nice to have another export format to populate the Home assistant Broadlink "Remote" entities from existing ProntoHEx parametric remotes

[https://www.home-assistant.io/integrations/broadlink#remote](url)
(see my example below)

With the Python ProntoHex to Base64 script + an Excel file, I managed to convert from the Pronto Hex to the format below
but It would be much faster if it could be done straight from IRScrutinizer =)

(essentially just a line by line export in this format: "Button name": "Broadlink_base64_code", )

```
  "data": {
    "AC TCL18K": {
      "OFF": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDAwMDAwmDAwMDAwMDAwMDAwMDAwMJgwmDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDCYMJgwmDAwMAAZxDQU=",
      "COOL16": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwmDCYMJgwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDCYMDAwmDAwMDAwmDAwMAAZxDQU=",
      "COOL17": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDCYMJgwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwmDAwMDAwmDAwMAAZxDQU=",
      "COOL18": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwmDAwMJgwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDAwMDAwmDAwMDAwmDAwMAAZxDQU=",
      "COOL19": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDAwMJgwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDAwMDAwmDAwMAAZxDQU=",
      "COOL20": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwmDCYMDAwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDCYMJgwMDAwMDAwmDAwMAAZxDQU=",
      "COOL21": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDCYMDAwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMJgwMDAwMDAwmDAwMAAZxDQU=",
      "COOL22": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwmDAwMDAwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDAwMJgwMDAwMDAwmDAwMAAZxDQU=",
      "COOL23": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDAwMDAwmDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMJgwMDAwMDAwmDAwMAAZxDQU=",
      "COOL24": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwmDCYMJgwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDCYMDAwMDAwMDAwmDAwMAAZxDQU=",
      "COOL25": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDAwMDAwmDAwMAAZxDQU=",
      "COOL26": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwmDAwMJgwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDAwMDAwMDAwMDAwmDAwMAAZxDQU=",
      "COOL27": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDAwMJgwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDAwMAAZxDQU=",
      "COOL16S": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwmDCYMJgwmDAwMDAwMDAwMDAwMDAwMJgwmDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDCYMDAwMDAwMDAwMDCYMAAZxDQU=",
      "COOL17S": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMJgwMDAwMDAwMDAwMDAwMDCYMJgwmDAwMDAwMDAwMDAwMDAwMJgwmDCYMDA [... cut ...]
      "HEAT26S": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwmDAwMJgwMDAwMDAwMDAwMDAwMDAwMJgwmDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwmDCYMJgwMDCYMJgwmDAwMAAZxDQU=",
      "HEAT27S": "JgDmAGczDCYMJgwMDAwMDAwmDAwMDAwmDCYMDAwmDAwMDAwmDCYMDAwmDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMDAwMDCYMDAwMDCYMDAwMDAwMDAwMDAwMDAwMDAwMJgwMDAwMDAwMDAwMDAwMDAwMJgwmDCYMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDCYMJgwMDCYMJgwmDAwMAAZxDQU="
    }
  }

```

bengtmartensson, 2023-09-01:
@Montreal666 : Thank you for your suggestions. I just checked in two fixed/new export formats, see #477. One does switches, the second one remotes, basically what you suggested above. Your feedback is welcome, in particular on usage of the remote export.

Montreal666, 2023-09-01:
Thank you! Much appreciated.
Timing is right as I'm still migrating to home assistant from another platform. will need such tools to help me along the way. 
Will keep you posted once I reach this part of the project. 
Cheers,

Montreal666, 2023-09-15:
> @Montreal666 : Thank you for your suggestions. I just checked in two fixed/new export formats, see #477. One does switches, the second one remotes, basically what you suggested above. Your feedback is welcome, in particular on usage of the remote export.

Thanks again,
Just tested the new "remote" export format; it does work, however for some reason it introduces a 5x repeat to the codes.

find attached my original RMDU and GIRR

for "power" the following Broadlink code is generated:

Jg**V**UAAABE4kRNBERERERERERERERERERERERNBE0ETQRNBE0ETQRNBERETQRERERETQRERERERERNBERETQRNBERETQRNBE0EQAEuwABE0URAA0F

if I double check here  https://dimagoltsman.github.io/Random-Broadlink-RM-Code-Generator/#
it does confirm 5 repeats.
0 repeats should be

Jg**B**UAAABE4kRNBERERERERERERERERERERERNBE0ETQRNBE0ETQRNBERETQRERERETQRERERERERNBERETQRNBERETQRNBE0EQAEuwABE0URAA0F

Can you confirm where in the process are the 5 repeats introduced.

Quite possible I am missing something, but obviously, I'd rather not have to edit every code to adjust repeats.

[!DEVICE CODES - BROADLINK.zip](https://github.com/bengtmartensson/IrScrutinizer/files/12614604/DEVICE.CODES.-.BROADLINK.zip)

bengtmartensson, 2023-09-17:
The fundamental problem is that the Broadlink (as used e.g. in Homeassistant) format is such a primitive format. It cannot represent an intro signal and a repeat signal. (Compare for example Pronto Hex, GlobalCache, Girs.)  Therefore, we have to generate a fix number of repetitions in an "export".

The "5" is admittedly arbitrarily chosen. It is defined on line 55 in the file broadlink.xsl, found in the exportformats.d directory, residing in the installation. (I apologize for not to giving the "magic number" a name, will fix.) You can of course change it in your installation. If you have a good suggestion for this, I would like to hear it.

Again, the root of the problem is the dumb design of Broadlink and its Homeassistant support.

Montreal666, 2023-09-17:
Thanks for the feedback. As long as I can manually change it, I should be OK. However, since this could be required at the button level, would it be feasible to add a "repeat" input field for each button in IRscrutinizer itself (when opening Girr files) so that the parameter could be set prior to export with a default value of 0. Also, on my previous platform (Homeseer), the developer added "pulse" adjustment (plus or minus 1,2,3,etc.)  which proved useful for some devices. Let me know what you think.

bengtmartensson, 2023-09-18:
> However, since this could be required at the button level, would it be feasible to add a "repeat" input field for each button in IRscrutinizer itself (when opening Girr files) so that the parameter could be set prior to export with a default value of 0.

Appears to be a misunderstanding. An "IrSignal" in IrScrutinizer consists of three "IrSequences", intro, repeat and ending; any can be empty. When transmitting, a variable number of repeats can be sent (Pane Hardware -> Count). Homeassistant can also send a variable number of repeats (that is probably your "pulse adjustment") -- just that the stupid Broadlink driver does not support it correctly. Possibly I should write a sane Broadlink driver for HA?

Montreal666, 2023-09-19:
Thanks, yes I was aware of the difference between of the repeat count within the "IrSequences" vs repeating the actual "IrSignal" (HA does offer the option). As you pointed out, changing line 55 in broadlink.xsl to zero does the trick. Was just wondering if this could be done at the button level directly in IRScrutinizer when importing a parametric remote Girr before exporting to Broadlink (vs changing it for all IrSignals by modifying broadlink.xsl )

As for modifying the pulse as mentioned above, it differs from IrSequence repeat or repeating the IrSignal;
it is defined as follows by the developer  ;

```
“repeat” is the number of times the code IR should be sent. A value of 0 is zero repeats. 1 is for one
repeat which means the code is sent twice.

“pulse” is the change in the number of pulses that represents each On/Off burst. This can be used to
tweak the timing of the IR signal for sensitive equipment. A value of -1 will reduce the number of pulses
at each transition by one. A value of 3 will increase the number of pulses by 3.
```
Again, this made a difference for some sensitive equipment I have here.

Thanks again for the tip regarding line 55 in broadlink.xsl 
I am not finished yet with converting all my IRcodes but it does the trick for now.
Will test the "sensitive" devices and report back.

Cheers,

Montreal666, 2023-09-19:
Did more tests today.
Fyi some of the conversion doesn't seem to work via IRScrutinizer.
One example for reference (remaining buttons behave the same way; conversion script works but not the conversion via IrScrutinizer)

**Working Pronto Hex from RMIR:**

0000 006D 0000 000E 000C 000C 000C 0018 0018 0018 000C 000C 000C 000C 0018 0018 0018 0018 000C 000C 0018 000C 000C 000C 000C 000C 000C 000C 000C 0018 000C 0D36

**Working Broadlink IrSignal from Python script** 
 https://onecompiler.com/python2/3zgv8s8n8 
(in this case the 5 IrSequence repeats are required, modified at line 83)

JgUeAAoKChQUFAoKCgoUFBQUCgoUCgoKCgoKCgoUCgALaA0FAAAAAAAAAAA=

**Non working Broadlink IrSignal from IrScrutinizer:**

JgUiAAoKChMTEwoKCgoTExMTCgoTCgoKCgoKCgoTCgANBQ== 

Let me know what you think.

Thanks!

Montreal666, 2023-09-19:
I'm done with the conversion of 19 devices.
Except for a single device ( ARRIS PVR - TDC-38 protocol ) where I had some issues (above) the rest didn't require further fine tuning other than changing the number of IrSequence repeats; generally zero, or a few repeats for some devices.
Therefore it would be useful to adjust the repeats from the GUI or maybe create 2 export templates (no repeats and 5 repeats maybe)

Thanks again for helping on this, should be useful for other HA users.

bengtmartensson, 2023-09-20:
The present Broadlink integration, as I said, I consider severely broken.
I plan to write my own "custom component" Broadlink integration, that uses (essentially?)  the same data format (Pronto Hex) as the "iTach" integration.  Hopefully soon...

Montreal666, 2023-09-20:
Thank you 👍.
Following.

Montreal666, 2023-09-20:
Just for future reference and if it can help in any way, here is the code the developer shared with me for the "pulse" component.
Thx.

```
  Private Function TweakCode(ByVal sHexCode As String, ByVal sRepeat As String, ByVal sPulse As String) As String

      Dim sb As New Text.StringBuilder

      Dim iDeltaPulse As Integer = CType(sPulse, Integer)

      Dim i As Integer = 0

      While i < sHexCode.Length

          If i = 0 Then '26

              sb.Append(sHexCode.Substring(i, 2))

          ElseIf i = 2 Then 'repeat

              sb.Append(CType(sRepeat, Integer).ToString("X2"))

          ElseIf i = 4 Then 'length lsb

              sb.Append(sHexCode.Substring(i, 2))

          ElseIf i = 6 Then 'length msb

              sb.Append(sHexCode.Substring(i, 2))

          Else 'pulse length on followed by off in pairs

              If iDeltaPulse = 0 Then

                  sb.Append(sHexCode.Substring(i, 2))

              Else

                  Dim sOnOff As String = sHexCode.Substring(i, 2)

                  Dim iPulse As Integer = Convert.ToInt32(sOnOff, 16)

                  If iPulse = 0 Then

                      If i < sHexCode.Length - 2 Then

                          If sHexCode.Substring(i + 2, 2) = "00" Then 'end of pulses with fill at the end

                              sb.Append(sHexCode.Substring(i, 2))

                          Else '2 byte pulse length

                              sOnOff = sHexCode.Substring(i + 2, 4)

                              iPulse = Convert.ToInt32(sOnOff, 16)

                              iPulse += iDeltaPulse

                              sb.Append("00")

                              Dim iPulseMSB As Integer = iPulse >> 8

                              Dim iPulseLSB As Integer = iPulse - (iPulseMSB << 8)

                              sb.Append(iPulseMSB.ToString("X2"))

                              sb.Append(iPulseLSB.ToString("X2"))

                              i += 4

                          End If

                      Else

                          Exit While

                      End If

                  Else

                      iPulse += iDeltaPulse

                      sb.Append(iPulse.ToString("X2"))

                  End If

              End If

          End If

          i += 2

      End While

      Return sb.ToString

  End Function
```

commit 3f3529a6 (2019-08-26): New export format Homeass-Broadlink. Resolves #312.