bengtmartensson/IrScrutinizer #317: IrScrutinizer_captured_signal not renderable, skipped

somejavadev, 2019-08-25:
Hi, I am using the latest version 2.0.1. I have an IR remote which I want to simply record and playback the individual buttons. I am making use of an IRToy with the latest firmware (222). When I record the signal I cannot export or scrutinize the signal. Scrutinising produces a `NumberFormatException` and exporting produces:

**Export Config:**

> begin remote   
	name		IrScrutinizerExport   
	flags		RAW_CODES   
	eps		30   
	aeps		100   
	frequency	40092   
	gap		   
	begin raw_codes   
		# IrScrutinizer_captured_signal not renderable, skipped   
	end raw_codes   
end remote   

**Screenshot of UI:**

[image]

**The stack trace of the number format exception is:**

`org.harctoolbox.ircore.InvalidArgumentException: java.lang.NumberFormatException: For input string: "{41493"
	at org.harctoolbox.ircore.MultilineIrSignalParser.toModulatedIrSequence(MultilineIrSignalParser.java:75)
	at org.harctoolbox.ircore.MultiParser.toModulatedIrSequence(MultiParser.java:89)
	at org.harctoolbox.analyze.RepeatFinderParser.toIrSignalAux(RepeatFinderParser.java:93)
	at org.harctoolbox.analyze.RepeatFinderParser.toIrSignalClean(RepeatFinderParser.java:79)
	at org.harctoolbox.irscrutinizer.InterpretString.interpretString(InterpretString.java:65)
	at org.harctoolbox.irscrutinizer.GuiMain.getCapturedIrSignal(GuiMain.java:1028)
	at `

**The signal being produced is:**

`Freq=41493Hz{41493,[+6947,-6947,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-1669,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-534,+534,-1669,+534,-1669,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-1669,+534,-534,+534,-534,+534,-534,+534,-534,+534,-1669,+534,-1669,+534,-534,+534,-1669,+534,-1669,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-534,+534,-1669,+534,-534,+534,-534,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-534,+534,-1669,+534,-1669,+534,-6947,+534,-1398078]}[][]`

Is there perhaps something I need to do/configure to be able to record and replay this signal?

bengtmartensson, 2019-08-25:
Hi,

I think it is [this problem](https://github.com/bengtmartensson/IrpTransmogrifier/issues/115), which is fixed in the development sources. Please try the [CI build](https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build). Another workaround is to select Options -> Output text format -> Pronto hex.

somejavadev, 2019-08-25:
Hi @bengtmartensson, thanks for the response. The CI build resolved the issue and is working fine. When I tried the Pronto Hex output format in version 2.0.1 I got a message mentioning erroneous data.

closed 2019-08-25 by somejavadev