bengtmartensson/IrScrutinizer #319: Internal error: IrSequence has odd length

clodoaldosouza, 2019-08-29:
Internal error: IrSequence has odd length ocurrs in version 2.0.0 and 2.0.1, but in version1.4.3 works fine.

Im using hardware IRWidget whith arduino software from ( https://gist.github.com/probonopd/5793692#file-sendandreceive-ino ) and baud rate changed to 115200.
 
The remote control is for a Air Conditioner Electrolux (same as Colix protocol).

There are something wrong?
 
Thanks in advance.

[image]
[image]

bengtmartensson, 2019-08-29:
By IrWidget the [device by Kevin Tiimmerman](http://www.compendiumarcana.com/irwidget/) is meant. Sometimes built by [Tommy Tyler](http://www.hifi-remote.com/forums/viewtopic.php?t=9405). It is supported directly IrScrutinizer, without any Arduino. So, how you are using with probonopd firmware is completely unclear. Please explain.

I suggest that you use [this firmware](https://github.com/bengtmartensson/AGirs) on the Arduino, see also http://www.harctoolbox.org/arduino_nano.html

Odd length sequences are evil, and fixed by Options -> Protocol Parameters -> Dummy Gap

clodoaldosouza, 2019-08-29:
Using AGirs, woks, but i decide to use old version 1.4.3.

Thanks.
[image]

closed 2019-08-29 by clodoaldosouza

bengtmartensson, 2019-08-29:
Nice to hear, but

> i decide to use old version 1.4.3.

may I ask why?

carloselectro, 2020-08-05:
I do have the same problem with version 2.2.6 and Irwidget, I was unable to capture my remotes until I have tried version 1.1.3 and it worked right away.