bengtmartensson/IrScrutinizer #320: IrScrutinizer-2.1.0.exe "Please install java before you install IrScrutinizer"

pico563, 2019-09-02:
IrScrutinizer-2.1.0.exe  refuse to install
**_"Please install java before you install IrScrutinizer"_**
 ---

1. java Version 8 update 221 (build 1.8.0_221-b11) already installed.
2. 
[image]

pico563, 2019-09-02:
what Java type and version should I install ?
thanks

bengtmartensson, 2019-09-02:
Thank you very much for the bug report. Sorry for that, the algorithm for detecting the existence of Java must have went wrong.

For a quick solution use [this new setup.exe](https://github.com/bengtmartensson/harctoolboxbundle/releases/download/Version-2.1.0/IrScrutinizer-2.1.0-no-javacheck.exe).

The long time solution is probably #283, not another half-working Java detection algorithm.

Again, thank you very much for the bug report.

pico563, 2019-09-02:
thank you for your good work.

closed 2019-09-02 by pico563