bengtmartensson/IrScrutinizer #321: Support irplus codefile Import

irplus-remote, 2019-09-10:
Hey, could you help me to implement import for irplus file? This would help me a lot with database maintenance. Currently i do it manually using the txt import. 

What needs to be done to add a new Tab within "Import" like LIRC ?

Basically what needs to be done is:

- Read line-by-line 
- Get the format from the format property -> pseudocode:  if contains NEC1 -> NEC, if contains PRONTO -> PRONTO if contains RAW -> RAW   (This should be enough protocols for the start
- strip everything off except the "label" which is used to name the button
- the code is the text inside <button></button> tag

NOTE: I've created this issue for when you have time to help, so i'm just asking kindly that one day in future it might be supported ;-) I'd like to help out as well, but i have not been able to build irScrutinizer myself.. Its seems so complicated to get it built. I had a lot of maven errors

Just one file for reference:
```xml
<irplus>
    <device manufacturer="2WIRE" model="2WIRE" columns="30" format="WINLIRC_NEC1" one-space="1662" one-pulse="585" zero-pulse="585" zero-space="538"
        header-pulse="9024" header-space="4462" bits="16" pre-bits="16">

        <button label="&#xf011;" backgroundColor="FFC84334" span="10">0x4F9 0x00FF</button>
        <button label="BACK" span="10">0x4F9 0x807F</button>
        <button label="MENU" span="10">0x4F9 0xC03F</button>

        <button label="VOL+" span="10">0x4F9 0xF00F</button>
        <button label="&#xf0d8;" span="10">0x4F9 0x40BF</button>
        <button label="&#xff04c;" span="10">0x4F9 0xD02F</button>

        <button label="&#xf0d9;" span="10">0x4F9 0xA05F</button>
        <button label="OK" span="10">0x4F9 0x609F</button>
        <button label="&#xf0da;" span="10">0x4F9 0xE01F</button>

        <button label="VOL-" span="10">0x4F9 0x30CF</button>
        <button label="&#xf0d7;" span="10">0x4F9 0x50AF</button>
        <button label="SEARCH" span="10">0x4F9 0x906F</button>

        <button label="&#xf04b;" alt="PLAY" span="15">0x4F9 0x708F</button>
        <button label="HOME" span="15">0x4F9 0xB04F</button>

        <button label="G" fillColor="FF1E7854" span="6">0x4F9 0x8877</button>
        <button label="B" fillColor="FF2A4C94" span="6">0x4F9 0x48B7</button>
        <button label="Y" fillColor="FFD6A81F" span="6">0x4F9 0x08F7</button>
        <button label="O" fillColor="FFFF7F00" span="6">0x4F9 0x20DF</button>
        <button label="V" fillColor="FF7F00FF" span="6">0x4F9 0xC837</button>
        
    </device>
</irplus>
```

closed 2019-09-10 by irplus-remote

bengtmartensson, 2019-09-10:
This a reasonable request, and it is the right place!

I have been planning to add extensible import formats for a long time. (Still to write the issue...) This should be basically a pre-processor -- an XSLT transformation -- that on the fly makes a Girr file out of the user's file. Recall, the export format are basically just XSLT transformations that makes a Girr file (in memory, as DOM, not on disk) into the user's format. 

So what I would recommend you to do is to write an XSLT script that makes a Girr file from an irplus file with raw or Pronto  signals. In the first step, you can then use that to convert your irplus files to Girr, and then import them. If/when I implement the preprocessor import, I will be happy to include your script. 

(You can also write an XSLT scipt that makes Lirc files out of the example above. But I am less inclined to incorporate a Lirc preprocessor.)

XSLT is reasonably easy to learn, and there are tons of books and support on the internet. For examples, see the export formats of IrScrutinizer. (Please use XSLT version 1 if you intend to later have it run within IrScrutinizer.)

irplus-remote, 2019-09-11:
Isn't this an overkill with all that XSLT stuff? Would it not be possible to add irplus import with hardcoded java code? (read line-by-line). At least it would be much easier for me to write a Java Parser

Pretty much like adding a new "IrplusImporter" Java file here:

/src/main/java/org/harctoolbox/irscrutinizer/importer/FileImporter.java

And adding a tab for it 
Bad paint-job incoming ;-)
[image]

I know my format is maybe not as much used as maybe pronto or lirc - but as least i think it is the most open and customizable IR format for android - Even if redesigning it today - id probably chose another format (JSON) as XML is giving me a lot of headache sometimes :)

bengtmartensson, 2019-09-11:
Sorry, I will not accept a fix like that. There are already too much clutter on the import pane.

The proposed solution #322 has several advantages, one of them is that a discussion over the importance of a format is not necessary any more.

sdaly31, 2023-07-04:
@irplus-remote, would you mind sharing how you were able to convert the irplus export file manually for use with IrScrutinizer? I am having trouble formatting it to import correctly. Alternatively, is there any way you know of to put this into an anymote import format?

@bengtmartensson What is the recommended way to import unsupported file types now in 2023? Is it still to use the text/CSV import?

Thanks!

`<irplus>
 <device manufacturer="" model="recieve" columns="4" format="UEI_LG_LEARNED" repeatheader="false">
  <button label="on" labelSize="12.0">1:[3, 0, 0, 0, 88, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, 48, -1, -1, 0, 0, 0, 50, 0, 0, 0, -85, -77, 32, -117, -74, -54, -1, -13, -104, 111, -122, -68, 55, 3, 108, 54, -127, 48, 80, -110, 73, 52, 86, 55, -128, 81, 82, 0, 29, -128, -94, 73, 44, 20, 73, 36, -109, 1, 68, -110, 73, 48, 20, 73, 36, -109, 64, 16, -128]</button>
  <button label="volume up" labelSize="12.0">1:[3, 0, 0, 0, 104, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, 64, -1, -1, 0, 0, 0, 66, 0, 0, 0, -85, -77, 72, 89, -99, 115, -104, -105, 6, -8, 108, 6, -68, 101, 69, -84, 78, -60, 72, 64, -61, 112, 54, -61, 104, 41, 65, -16, 19, 0, -8, 12, 10, 9, 36, -110, 73, 43, 96, -54, 101, 57, 47, 68, 15, 33, 68, 3, 106, -56, -80, -128, 10, 64, 36, 104, -84, -32, 48, 0, 19, 0, 20, 16, -128]</button>
  <button label="volume down" labelSize="12.0">1:[3, 0, 0, 0, 56, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, 16, -1, -1, 0, 0, 0, 18, 0, 0, 0, -84, 51, 12, 77, -62, -125, 60, 15, -128, 56, 16, -104, -48, 0, 20, 16, -128]</button>
  <button label="Off" labelSize="12.0">1:[3, 0, 0, 0, 72, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, 32, -1, -1, 0, 0, 0, 34, 0, 0, 0, -83, 67, 20, 74, 116, -54, 100, 55, -125, 84, 27, -63, -88, 80, -110, 73, 46, 57, 85, 78, 0, 1, 76, 20, -117, 73, 40, -110, 73, -107, 64, 16, 64]</button>
  <button label="CD" labelSize="12.0">1:[3, 0, 0, 0, -71, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, -111, -1, -1, 0, 0, 0, -109, 0, 0, 0, -85, -77, 45, -111, 8, -49, 52, -51, -112, 74, -29, -120, 58, 71, 122, -77, -104, 111, -122, -68, 55, 3, 108, 54, -118, 9, 36, -110, 73, 51, 79, 15, 115, 18, 114, -78, -110, 50, 82, 18, -64, 12, -3, 28, 4, 72, -64, 0, 1, 0, 68, -116, 0, 0, 20, 0, 4, -116, 0, 0, 16, 0, 4, -116, 0, 0, 16, 0, 4, -116, 0, 0, 24, 4, 72, -64, 0, 1, 0, 68, -116, 0, 0, 16, 4, 72, -64, 0, 1, -64, 0, 72, -64, 0, 1, 0, 0, 72, -64, 0, 1, 0, 0, 72, -64, 0, 1, 0, 4, -116, 0, 0, 16, 0, 4, -116, 0, 0, 32, 4, 72, -64, 0, 1, 0, 68, -116, 0, 0, 36, 0, 4, -116, 0, 0, 16, 0, 4, -116, 0, 0, 16, 42, 64, 16, -88]</button>
  <button label="dvd" labelSize="12.0">1:[3, 0, 0, 0, -67, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, -107, -1, -1, 0, 0, 0, -105, 0, 0, 0, -85, 51, 53, 87, 27, -115, 28, 74, -18, 74, 47, -56, -26, 89, -72, -92, 69, 36, 82, 67, 108, 54, -61, 108, 27, -127, -76, -80, -110, 73, 36, -110, 90, 114, -13, 33, -22, 107, 112, -16, 112, -80, 49, 49, 64, 25, -6, 56, 9, 17, -111, 17, 33, 17, 41, 17, 19, 0, -111, 25, 17, 18, 17, 18, -111, 17, 48, 9, 17, -111, 17, 33, 17, 41, 17, 19, -128, -111, 10, -111, 17, 33, 17, 41, 17, 19, 0, -111, 10, -111, 17, 33, 17, 41, 17, 19, 0, -111, 10, -111, 17, 33, 17, 41, 17, 20, 0, -111, 25, 17, 18, 17, 41, 17, 20, -128, -111, 25, 17, 18, 17, 18, -111, 17, 80, 9, 16, -87, 17, 18, 17, 18, -111, 17, 48, 9, 16, -87, 17, 18, 17, 18, -111, 17, 48, 9, 92, -128, 16, -88]</button>
  <button label="vcr" labelSize="12.0">1:[3, 0, 0, 0, 66, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, 26, -1, -1, 0, 0, 0, 28, 0, 0, 0, -83, 67, 16, 74, -106, 6, -16, 106, -125, 120, 53, 68, 9, 36, -72, 99, 69, 48, 0, 27, 0, -90, 0, 0, 64, 16, 64]</button>
  <button label="CD 1" labelSize="12.0">1:[3, 0, 0, 0, 81, 21, -95, 13, -16, 2, 79, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 2, 0, 0, 0, 0, 0, 0, 0, 41, -1, -1, 0, 0, 0, 43, 0, 0, 0, -85, -77, 32, -87, 30, 10, -27, 51, -104, 111, -122, -68, 55, 3, 108, 54, -128, -8, 96, -110, 73, 42, -58, -122, -22, -118, 48, 10, 64, 2, -72, 37, 48, 0, 16, 80, 74, 96, 0, 48, 16, -128]</button>
 </device>
</irplus>`

bengtmartensson, 2023-07-05:
@sdaly31: What format is this? format="UEI_LG_LEARNED" is not something that is generally understood, and thus not suitable as an export format. Please try to use pronto hex.

> What is the recommended way to import unsupported file types now in 2023?

Try to write an input filer that transforms the input format to Girr. Either as an XSLT script (see earlier in this thread) or as a shell script that reads stdin and writes stdout. That way you can easily transform your input to Girr, which is the native format of IrScrutinizer. A later version may support involing the transformer automatically, see #322.