bengtmartensson/IrScrutinizer #326: Undecodable signal, ignored

bakdamir, 2019-10-16:
LocalSerialPortBuffered.readString: received "f=38461 +4466 -4507 +562 -564 +560 -1689 +562 -1691 +560 -1694 +560 -565 +560 -565 +560 -566 +560 -566 +562 -561 +562 -1689 +562 -1691 +560 -1693 +560 -565 +560 -565 +560 -565 +560 -580 +562 -563 +562 -561 +562 -1691 +560 -1693 +562 -565 +560 -565 +560 -565 +560 -563 +560 -1689 +562 -1694 +560 -565 +560 -563 +560 -1691 +560 -1691 +560 -1689 +562 -1693 +562 -46198 +4469 -4502 +560 -1687 +560 -65535 +4469 -4504 +560 -1687 +562 -65535 +4469 -4502 +560 -1687 +560 -65535 +4469 -4501 +562 -1685 +562 -65535 +4469 -4502 +560 -1685 +562 -65535 +4469 -4501 +560 -1687 +562 -65535 +4468 -4501 +562 -1687 +560 -65535 +4469 -4502 +560 -1687 +562 -65535 +4467 -4502 +562 -1685 +562 -65535"
Undecodable signal, ignored

bengtmartensson, 2019-10-16:
Please tell us in detail  what you are trying to do, what version you are using, hardware used, and what you expect.

bakdamir, 2019-10-16:
Trying to read the remote.
[image]

bakdamir, 2019-10-16:
[image]

bakdamir, 2019-10-16:
trying to read the remote

bengtmartensson, 2019-10-16:
Does it help selecting  OPtions -> Invoke repeat finder, Options -> Invoke signal cleaner?

bakdamir, 2019-10-17:
Helps with scrutinize signal,at scrutinixe remote without changes

bengtmartensson, 2019-10-17:
Good news: I have been able to reproduce!

bengtmartensson, 2019-10-17:
Turned out that it was much more "interesting" than I thought. Thank you very much for the report!!

Please try: https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build; it should fix the problem.

I would like you to test both with and without "Invoke repeatfinder" enabled.

bakdamir, 2019-10-17:
Thank you. I will definitely test and let you know.

bakdamir, 2019-10-17:
Invoke repeat finder enabled
[image]
Invoke repeat finder disabled
[image]

bengtmartensson, 2019-10-18:
That looks quite good. Can we consider this fixed and close the issue?

bakdamir, 2019-10-18:
Yes.

closed 2019-10-18 by bakdamir