bengtmartensson/IrScrutinizer #327: Using IrpTransmogrifier to generate lists of Pronto HEX codes

IanSav, 2019-10-23:
I am writing in search of some assistance with the latest generation of IrScrutinizer (Version 2.1.0).  I have tried to find my own answers in the data available on the harctoolbox website but so far I have not been successful.

I used to use the option of the IrpMaster function to generate lists of Pronto HEX codes.  The IrpMaster function has now been replaced with the IrpTransmogrifier function.  Unfortunately the documentation provided is not helping me to map the old features onto the new code.  (This is a case where a few syntax examples would be most helpful.  ;) )

For my current needs I used to use the following command to generate a list of Pronto HEX code for a remote control:
	`IrpMaster --decodeir --pronto NEC1 39 205 0:255`
This equates to:
	`Java -splash: "-Djava.library.path=C:\Program Files
(x86)\IrScrutinizer\Windows-x86"  -jar "C:\Program Files
(x86)\IrScrutinizer\IrScrutinizer.jar" --irpmaster -c " C:\Program Files
(x86)\IrScrutinizer\IrpProtocols.xml"  --decodeir --pronto NEC1 39 205 0:255`

This would generate a list of all the Pronto HEX codes for functions F ranging from 0  to 255 for a NEC1 type device with a device code of D=39, S=205.  I can not find the equivalent range functionality with the new IrpTransmogrifier function.

I have worked out the syntax for a single Pronto HEX code to be:
	`Java -cp "C:\Program Files (x86)\IrScrutinizer\IrScrutinizer.jar"
org.harctoolbox.irp.IrpTransmogrifier render --pronto --nameengine
"{name=NEC1, D=39, S=205, F=0}" NEC1`
but this simply generates a single code for the nominated F value.

Is there a way to replicate the original functionality of IrpMaster to output a range of Pronto HEX codes for a range of F values?  I have tried invoking the IrpTransmogrifier once for each F value but this takes ***significantly*** longer to process than the previous code that could generate all 256 combinations in a single run.

bengtmartensson, 2019-10-23:
You can fill in, e.g. 0:255 as "F" on the "Generate" pane and then export, as "Text" or another export format, (like Girr). It *is* [documented](http://www.harctoolbox.org/IrScrutinizer.html#Accessing+a+number+of+different+parameter+values) , although I agree that it is quite "concise". 

There is presently no way to do this from the command line, but it is planned, see #26 .

There is a wrapper for IrpTransmogrifier as C:\Program Files
 (x86)\IrScrutinizer\IrpTransmogrifier.bat that you can use, like 

    % irptransmogrifier render --pronto --nameengine "{D=39, S=205, F=0}" NEC1

provided that you let the setup file modify your path. Otherwise you can call it using the full path.

BTW, installing both version 1.4.3. and the latest one (in different install directories) can be a good idea.

IanSav, 2019-10-23:
Unfortunately I need the command line version as this task is part of a pipeline process.

I didn't realise that both the old and new version can coexist.  I will give that a try to bridge the gap until IrpTransmogrifier gains the command line options I need.

bengtmartensson, 2019-10-24:
> Unfortunately I need the command line version as this task is part of a pipeline process.

Ok, then it is surely the best short-time solution to install IrScrutinizer-1.4.3 (or [IrpMaster-1.4.2](https://github.com/bengtmartensson/IrpMaster/releases) (in a separate directory).

There *will* be a possibility to do this with the current version, but not today, not this week, not this month...

IanSav, 2019-10-25:
Thank you for the advice.

closed 2019-10-29 by bengtmartensson