bengtmartensson/IrScrutinizer #333: Undecodable signal, ignored

bakdamir, 2019-11-09:
Freq=38461Hz[+854,-872,+854,-874,+1733,-872,+854,-872,+854,-872,+854,-870,+856,-871,+856,-870,+854,-1721,+854,-872,+1735,-870,+856,-65535,+856,-872,+854,-873,+1732,-873,+854,-870,+854,-872,+854,-872,+854,-872,+855,-872,+854,-1719,+854,-874,+1733,-870,+856,-65535,+854,-874,+852,-875,+1733,-872,+854,-870,+854,-872,+854,-873,+854,-872,+854,-872,+854,-1721,+854,-872,+1733,-872,+854,-65535,+854,-872,+854,-872,+1735,-872,+854,-870,+856,-870,+854,-872,+855,-872,+854,-872,+854,-1721,+854,-872,+1732,-872,+854,-65535][][]

bakdamir, 2019-11-09:
[image]
[image]

bengtmartensson, 2019-11-09:
Working for me

[image]

(It turns on/off a Philips TV :-)).

Make sure you really are using the latest version; then reset properties (File -> Set properties to default; restart).

bakdamir, 2019-11-09:
Without changes

bengtmartensson, 2019-11-10:
Try today's ci-build.

bengtmartensson, 2019-11-13:
Still waiting for feedback.

I cannot reproduce and honestly have no idea what the problem is. Will close shortly unless I get useful feedback.

bakdamir, 2019-11-13:
Let's close it for now

closed 2019-11-13 by bakdamir