bengtmartensson/IrScrutinizer #337: "Undecodable signal, ignored"

DanielTheGreat, 2019-11-17:
When I tried to 'scrutinize remote' IR signals from a Pioneer GR-777 remote, IRScrutinizer (v2.1.0) said: "Undecodable signal, ignored".  But when I used 'scrutinize signal', it had no trouble receiving and displaying the code details (see attached screen-shot).  Why does 'scrutinize remote' not work for this remote?
[image]

bengtmartensson, 2019-11-17:
Do you expect me to retype your signal or run your piccy through OCR?

DanielTheGreat, 2019-11-17:
Sorry, I think you've mentioned this before, but I don't use this forum often enough for my 75-YO brain to remember. :-(
Here's an export of a not-decoded signal.

[girr_2019-11-17_18-51-35.txt](https://github.com/bengtmartensson/harctoolboxbundle/files/3855340/girr_2019-11-17_18-51-35.txt)

bengtmartensson, 2019-11-17:
You screendump in the first message show that it decodes, and I get the same result. 
 
Possibly #326 is of relevance. Try the current [CI build](https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build) (where that problem is fixed).

PS. Preferred extension for Girr files is `.girr` or `.xml`.

DanielTheGreat, 2019-11-17:
1. I downloaded, installed and tried IrScrutinizer-2.2.0-3d50bdc-SNAPSHOT.exe, but it still doesn't recognise the Pioneer remote in 'Scrutinize remote' mode; not in Parametric mode ('Undecodable signal, ignored'), at least.  But it does record it in Raw mode (see screen-shot).
[image]
2. "PS. Preferred extension for Girr files is .girr" – please tell that to GitHub, which rejected my .girr file upload!  I maybe could have zipped it, but it was far quicker to just rename the extension, knowing you would recognise the filetype from its name, and rename it back if desired.

DanielTheGreat, 2019-11-17:
PS: Your latest snapshot was flagged as containing a virus by three engines, including two well-known ones!
[image]

bengtmartensson, 2019-11-17:
> I downloaded, installed and tried IrScrutinizer-2.2.0-3d50bdc-SNAPSHOT.exe, but it still doesn't recognise the Pioneer remote in 'Scrutinize remote' mode

Under investigation

> PS: Your latest snapshot was flagged as containing a virus by three engines, including two well-known ones!

Do the following: Verify that the download is not corrupted (It could be the case that your downloader is infected, and infects, or otherwise mutliates downloads). For this, download the checksums (*.md5 and *.sha1) and verify that the download is OK (see http://www.hifi-remote.com/forums/viewtopic.php?t=14986&start=128 ), Assuming it is OK, please report false positives to your virus checker provider.

bengtmartensson, 2019-11-20:
Replace by #344.

closed 2019-11-20 by bengtmartensson