bengtmartensson/IrScrutinizer #338: Wrong export path

DanielTheGreat, 2019-11-17:
I entered a path in the 'Export dir.' field, but IRScrutinizer (v2.1.0) totally ignored that path and created the file in the default directory (see screen-shot).
[image]

bengtmartensson, 2019-11-17:
You will have to trigger the action of the text file, for example by pressing return, just typing is not enough.

PS. Please use a describing, neutral language and avoid emotionally colored terms. "Totally ignored" implies deliberate evil.

DanielTheGreat, 2019-11-17:
Sorry about the anthropomorphism!

If this bug (having to hit <Enter> after entering a path) can't be fixed, then please add a warning in the Export pane that this non-standard behaviour is required.

closed 2019-11-17 by bengtmartensson