bengtmartensson/IrScrutinizer #339: Formats of IR codes

DanielTheGreat, 2019-11-17:
Programs I use to send IR commands from a PC to my equipment (see ) expect the codes to be a single hex number for each command, like this:
#define PG_PRE_PRG	0xCC335FA0L		// Preset/Program curves
(Obviously the program needs to know the pulse timings too, in order to send commands.)
But IRScrutinizer neither decodes nor sends commands in such a simple format.
This makes it difficult to 'explore' a piece of equipment by entering (in IRScrutinizer) codes not already included in a remote in the hope of finding 'hidden' equipment commands to perform functions available on the front panel but not on the remote.
It would be very useful if IRScrutinizer, after determining the timings from a sample signal from an existing remote, allowed the **simple** entry (e.g. as a whole hex number representing a code value, as in the above example) of 'new' codes (not already available in that remote) to transmit and thus explore for hidden commands.
Perhaps there is already a simple way for doing this, but if so, it is not well documented.

bengtmartensson, 2019-11-17:
See the protocol NEC-Shirriff and NEC-Schirriff-32, and compare to NEC1. These are compatible with the widely used [IRremote](https://github.com/z3t0/Arduino-IRremote) library by Ken Shirriff. Otherwise, you will have to ask the authors of the

> Programs I use to send IR commands from a PC to my equipment

DanielTheGreat, 2019-11-17:
I Googled for 'protocol NEC-Shirriff ' and a few other search-term variants, but couldn't find any definition document for those protocols.  I found Ken Shirriff's Arduino project for an IR Rx and Tx, which presumably uses those protocols, but I haven't studied the code yet (I consider that to be the hard and incorrect way to learn a protocol – a definition document is far more professional).
But IRScrutinizer doesn't include export options for NEC-Shirriff and NEC-Schirriff-32 protocols, so I'm still stumped unless I switch to using Ken's apps instead of yours.  I haven't tried them yet, but suspect they won't be as universal or flexible as IRScrutinizer.
Incidentally, when posing this initial question, I forgot to locate and paste the link to my video showing how I use a PC to control various items of (disparate) audio equipment via IR signals, to illustrate how I use the described simple way of sending codes with just two arguments, a hex code value and a protocol number.  Here's that link:
[https://youtu.be/xo-sODkjKOk](url)

bengtmartensson, 2019-11-17:
NEC-Shirriff is not an official notation, it is just my name for what IRremote calls "NEC" -- to distinguish from the real NEC protocols. They are found in the protocol data base of IrScrutinizer, not on  Google. This is an exampe of a "dumb" interpretation of protocol -- name + payload data, just what you asked for.

> But IRScrutinizer doesn't include export options for NEC-Shirriff and NEC-Schirriff-32 protocols

Wrong.

DanielTheGreat, 2019-11-17:
OK, I'll bite – how does one export in a Shirriff NEC format when presented with only the export format options shown in this screen-shot?
[image]

bengtmartensson, 2019-11-17:
Just (for example) use "Scrutijnize parametric remote", and fill in "NEC-Shirriff" manually as protocol, and type something like data=0x123456A9 as "Misc params". Then export parametric remote.

DanielTheGreat, 2019-11-18:
Bengt,

I'm probably a bit dense, but I have no idea what you're asking me to do here!

If I select 'Scrutinize remote' and have the (default) 'Parametric Remote' tab selected, I cannot manually enter text anywhere except in the un-named field near the top (next to the 'Export' button).  If I enter 'NEC-Shirriff' there (without the quotes) and hit <Enter>, nothing changes, and I still see no way to enter manual data for 'Misc. params'.

Just as an experiment (to see if you had missed an important instruction), I captured a command from a recognised (Denon) remote I had handy.  I was then able to click on the empty 'Misc. params' data filed and enter your suggested string 'data=0x123456A9' (without quotes), but that generated a flood of Java errors (see screen-shot and text copy of error messages).
[image]
[Java_errors.txt](https://github.com/bengtmartensson/harctoolboxbundle/files/3856518/Java_errors.txt)

Apart from my inability to execute your instructions, I'm not sure what the point is anyway.  If I have to manually enter the remote codes as hex numbers before exporting them, I achieve nothing, because I could manually type into my C header file in the first place, where I ultimately want them.

Unless IRScrutinizer can decode a remote's whole hex numbers (one per command) and export them as whole hex numbers, it is of little use to me.  Likewise, for seeking 'hidden' equipment commands, I nwant to be able to enter whole hex commands and transmit them.

bengtmartensson, 2019-11-18:
> I cannot manually enter text anywhere except in the un-named field near the top (next to the 'Export' button). 

You create an empty row by right mouse -> Add empty signal, then you can edit that one. 

> that generated a flood of Java errors (see screen-shot and text copy of error messages).

Ok, I see that it does not take the 0x prefix (#343). You have to enter decimal numbers.

It seems like you want to create "Shirriff-style" versions of the protocols in IrpProtocols.xml. This can "simply" be done by creating and using your own version of IrpProtocols.xml where you define your own Shirriff style protocols, and delete the existing (or use prefer-over). I do not consider this a sane idea; you are on your own.

DanielTheGreat, 2019-11-20:
In the scant IRScrutinizer documentation I have downloaded, there is nothing about how IrpProtocols.xml is used in IRScrutinizer, nor (more importantly) the meaning of its entries and how to make new ones to display and/or transmit codes in a sensible (hex number) format.  I see it already has entries for NEC-Shirriff and NEC-Shirriff-32.  Are they not immediately usable for this purpose?

Having to enter decimal numbers for what is transmitted as a binary code seems like a backward step!  Hex would be far more appropriate.

bengtmartensson, 2019-11-20:
> In the scant IRScrutinizer documentation

Please adjust your tone. It is not exactly the first time ...

> there is nothing about how IrpProtocols.xml is used in IRScrutinizer, nor (more importantly) the meaning of its entries and how to make new ones to display and/or transmit codes in a sensible (hex [...]

This is covered in the IrpTransmogrifier documentation.

> Having to enter decimal numbers for what is transmitted as a binary code  seems like a backward step!

See #343.

I consider questions as answered and close therefore,.

closed 2019-11-20 by bengtmartensson