bengtmartensson/IrScrutinizer #356: Support Tasmota devices

probonopd, 2020-01-24:
[Tasmota](https://github.com/arendst/Tasmota) is a very popular open source firmware for ESP8266 based devices. It is used in many home automation systems.

It can send and receive IR codes via WLAN.
https://github.com/arendst/Tasmota/wiki/IRSend-RAW-Encoding has the technical details.

It would be cool if Tasmota devices could be supported in IrScrutinizer.

probonopd, 2020-01-24:
Also see

* https://github.com/arendst/Tasmota/wiki/Tasmota-IR
* https://github.com/arendst/Tasmota/issues/2116#issuecomment-440716483

bengtmartensson, 2020-01-25:
Thank you very much for the suggestion. (Also, nice to hear from you!)

IrScrutinizer is mainly a tool for analyzing and collecting IrSignals, and to help setting up home automation systems. The need to support a large number of input/output devices is therefore somewhat secondary, in particular since (currently) the addition of new hardware support to the GUI is not quite streamlined. However, for ["IrScrutinizer as server", JGirs](https://github.com/bengtmartensson/JGirs) it is certainly desirable. See [HarcHardware](https://github.com/bengtmartensson/HarcHardware/issues/19) for implementation. Help most definitely welcome.

Having said that, if the goal is to integrate IrScrutinizer (and/or JGirs), it may be a better idea to teach Tasmota [Girs](http://harctoolbox.org/Girs.html) over TCP-sockets. (See also [AGirs](https://github.com/bengtmartensson/AGirs) for an example.) That would mean no changes  to IrScrutinizer would be necessary, and, as a free bonus, Lirc support would come in for free. (There is a Girs driver in the current officual Lirc; happens to be written by yours truly). 

Help most definitely welcome.

probonopd, 2020-01-26:
Hi @bengtmartensson I understand that implementing a GIRS server in LIRC would be technically preferable, however since Tasmota is trying to cram as many different sensors and protocols into one firmware as possible I assume it is very unlikely that they will add another IR handling library in addition to what they are already using there (and I further assume that they have lots of existing users which are depending on the existing IR functionality to stay as it is). 
Tasmota is designed to run on tiny, space-restricted devices. In contrast, in IRScrutinizer and the libraries it is using, there is no such space constraint. Hence I think it would be possible and worthwhile to add it there, especially since Tasmota can handle Raw codes sent over HTTP. The format is only slightly different from what IRScrutinizer is already capable of.

The advantage would be that IRScrutinizer would immediately become __the__ tool for Tasmoa users for all things IR, and they have a very large user base in home automation, usually tech-savvy home automation enthusiasts. I think it would be a great multiplicator effect for this project.

probonopd, 2020-03-29:
After some experimentation, this seems really easy to do.

Let's say you have the following raw timings:

```
+9041 -4507 +573 -573 +573 -573 +573 -573 +573 -573 +573 -573 +573 -573 +573 -573 +573 -573 +573 -1694 +573 -1694 +573 -1694 +573 -1694 +573 -1694 +573 -1694 +573 -1694 +573 -1694 +573 -573 +573 -1694 +573 -573 +573 -1694 +573 -573 +573 -573 +573 -573 +573 -573 +573 -1694 +573 -573 +573 -1694 +573 -573 +573 -1694 +573 -1694 +573 -1694 +573 -1694 +573 -39785
+9041 -2267 +573 -96193
```

Replace the ` +` and ` -` with `,` and add `irsend 0,` in front of it:

```
irsend 0,9041,4507,573,573,573,573,573,573,573,573,573,573,573,573,573,573,573,573,573,1694,573,1694,573,1694,573,1694,573,1694,573,1694,573,1694,573,1694,573,573,573,1694,573,573,573,1694,573,573,573,573,573,573,573,573,573,1694,573,573,573,1694,573,573,573,1694,573,1694,573,1694,573,1694,573,39785,9041,2267,573,96193
```

Then POST it to the device like so:

http://192.168.0.27/cm?cmnd=irsend%200,9041,4507,573,573,573,573,573,573,573,573,573,573,573,573,573,573,573,573,573,1694,573,1694,573,1694,573,1694,573,1694,573,1694,573,1694,573,1694,573,573,573,1694,573,573,573,1694,573,573,573,573,573,573,573,573,573,1694,573,573,573,1694,573,573,573,1694,573,1694,573,1694,573,1694,573,39785,9041,2267,573,96193

Works for me.

I would be really happy if IrScrutinizer could send codes to Tasmota devices this way. 

User case:
As a home automation user, I already have over 10 devices running Tasmota all over the place. It would be cool if I could just add an IR LED to all of them, and have them act as IR blasters controlled by IrScrutinizer.

probonopd, 2021-05-30:
Tasmota now (also) uses a new IR Raw compact encoding: https://tasmota.github.io/docs/IRSend-RAW-Encoding/