bengtmartensson/IrScrutinizer #384: JVC-48 and JVC-56 problems

kdschlosser, 2020-05-26:
This is what I think I know

The JVC-48 protocol is derived from the Kaseikyo protocol and
the JVC-56 protocol is derived from the Kaseikyo-56 protocol.

for both JVC protocols the M and N parameters must be set to 3 and 1. This is what identifies JVC as the manufacturer. If the M and the N are set to 3 and 1 then the protocol decoding should be finished up by the appropriate JVC decoder.

This is not happening when I scrutinize a JVC-48 or a JVC-56 that is generated by IRScrutinizer

I generated this JVC-56 code using the following parameters.
D=10
S=25
F=50
X=255

it returned the following code
```
0000 0070 0000 003A 0080 0040 0010 0030 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0030 0010 0030 0010 0010 0010 0010 0010 0010 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0010 0010 0030 0010 0010 0010 0010 0010 0030 0010 0030 0010 0010 0010 0010 0010 0010 0010 0030 0010 0030 0010 0030 0010 0030 0010 0010 0010 0030 0010 0030 0010 0ACD
```
when i put that code into the Scrutinize This is what it returns.
```
protocol = Gap-432-1297-56?, device = 3, subdevice = 1, obc = 222, misc = 3.1.A.19.FF.32.DE
```

and for JVC-48 I used these parameters
D=10
S=25
F=50

and this is the code that got generated
```
0000 0070 0000 0032 0080 0040 0010 0030 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0030 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0030 0010 0030 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0ACD
```
and when run through the scrutinizer
```
protocol = Gap-432-1297-48?, device = 3, subdevice = 1, obc = 33, misc = 3.1.A.19.32.21
```

I am not sure of the purpose to the G parameter for Kaseikyo and the Kaseikyo-56 protocol with relation to the JVC protocols. It appears as this parameter is unused in the JVC protocols. Yet when I generate codes in either JVC-56 or JVC-48 the G portion of the code changes value.

Now if I enter the same parameters I used in the JVC-48 parameters when making a Kaseikyo code and filling in the M (3), N(1) and G(33) parameters that are gotten from the JVC-48 protocol the generated Kaseikyo code should be identical, they are not.

So there is something wrong with the generator for the following protocols JVC-48, JVC-56, Kaseikyo and Kaseikyo-56. and the decoder is non functioning for JVC-48 and JVC-56 when using a generated code. Now this can be because the generated code is incorrect or there is actually something wrong with the decoder, but as a rule of thumb if a Kaseikyo or Kaseikyo-56 code has an M of 3 and an N of 1 it is a JVC code.

From looking at the binary data for the codes after stripping of the lead-in and lead-out and from another project that has written an ir decoder for Kaseikyo is that there may be a missing parameter for Kaseikyo and Kaseikyo-56.

Here is the RLC for a Kaseikyo-56 code

```
3458 , -1729 , # lead-in (header)

# data block ===================================================

# M (bits 0-7)
432 , -1296 ,
432 , -1296 ,
432 , -432 ,
432 , -1296 ,
432 , -432 ,
432 , -432 ,
432 , -1296 ,
432 , -432 ,

# N (bits 8-15)
432 , -432 ,
432 , -432 ,
432 , -1296 ,
432 , -432 ,
432 , -432 ,
432 , -1296 ,
432 , -1296 ,
432 , -432 ,

# G1 (bits 16-19)
432 , -1296 ,
432 , -432 ,
432 , -1296 ,
432 , -1296 ,

# Device (bits 20-23)
432 , -432 ,
432 , -1296 ,
432 , -432 ,
432 , -1296 ,

# sub device (bits 24-31)
432 , -1296 ,
432 , -432 ,
432 , -432 ,
432 , -1296 ,
432 , -1296 ,
432 , -432 ,
432 , -432 ,
432 , -432 ,

# E (bits 32-39)
432 , -432 ,
432 , -1296 ,
432 , -1296 ,
432 , -432 ,
432 , -1296 ,
432 , -432 ,
432 , -432 ,
432 , -1296 ,

# Function (bits 40-47)
432 , -432 ,
432 , -1296 ,
432 , -432 ,
432 , -432 ,
432 , -1296 ,
432 , -1296 ,
432 , -432 ,
432 , -432 ,

# G2 (bits 48-55)
432 , -1296 ,
432 , -432 ,
432 , -1296 ,
432 , -1296 ,
432 , -1296 ,
432 , -1296 ,
432 , -1296 ,
432 , -432 ,

# ==========================================================

432 , -74709 , # lead-out (footer)
```

the E parameter does not exist for the Kaseikyo protocol and as a result the code would be 8 bits smaller but otherwise identical.

The JVC-48 protocol and the JVC-56 protocol has a D parameter that is 1 byte in size (0-255) it does this by combining the G1 and D parameters together to make the 8 bits needed.

I could be incorrect in what I am seeing because I am using the codes that are generated by IRScrutinizer which could be generating the codes incorrectly. Or I could be misunderstanding the whole thing in it's entirety.

bengtmartensson, 2020-05-26:
Obviously, you are using IrScrutinizer version 1.*.*. The current version has a completely different rendering and decoding engine.  (The first signal decodes!)

Please use the current version 2.2.6.

kdschlosser, 2020-05-27:
you are correct good sir. I do not know why I had an old version installed. Sorry for disturbing you with my bone headed mistake.. LOL

All is good now.

closed 2020-05-27 by kdschlosser