bengtmartensson/IrScrutinizer #399: Remove Lirc/CCF as sending device

bengtmartensson, 2020-08-14:
See http://www.harctoolbox.org/IrScrutinizer.html#The+%22Lirc%22+pane . Anyone using this?

For the other half of Lirc support, see #128

closed 2021-07-04 by bengtmartensson

lbschenkel, 2022-12-25:
Sorry for not not saying anything sooner, but I just noticed this now. I was one of your users who was using this functionality. I implemented a bridge that talks the LIRC network protocol, so then Broadlink hardware could be used as a sending device in IrScrutinizer (https://github.com/lbschenkel/broadlink-bridge).

Is there any hope for restoring this?

bengtmartensson, 2022-12-25:
The reason that the functionality was removed was that the Lirc send named command ("irsend") stuff was moved elsewhere, and it was not clear if there was any serious usage.  The patch on my web page has not been maintained, and almost surely does not apply any more. (I also expect complaints from IrToy (non-Linux) users, but somehow it has not arrived yet...) 

So, if I understand you correctly, your basic requirement is to use IrScrutinizer with Broadlink for sending, rite? I have had [this](https://github.com/bengtmartensson/HarcHardware/issues/6) on my todo list for a long time (2018 to be exact), wouldn't that be a better solution?  Another possibility would be to implement the [Girs](http://www.harctoolbox.org/girs.html) command language in your bridge, you basically send a command like `send 1 38400 68 4 0 9024 4512 564 564 564 1692 564 564 564 1692 564 1692 564 1692 564 1692 564 564 564 1692 564 564 564 1692 564 564 564 564 564 564 564 564 564 1692 564 1692 564 564 564 1692 564 1692 564 1692 564 564 564 564 564 564 564 564 564 1692 564 564 564 564 564 564 564 1692 564 1692 564 1692 564 39756 9024 2256 564 96156` to the thing. This is the command language of [this device](http://www.harctoolbox.org/arduino_nano.html); I have also used it over Ethernet TCP sockets. Lircd supports it through the [Girs driver](https://www.lirc.org/html/girs.html) (both serial and TCP sockets) -- so you would actually provide Broadlink access for Lirc as an free extra. (Not that I care for Lirc...)

Let me know what you think. I look forward to hear from you.

lbschenkel, 2022-12-27:
The primary purpose for the bridge was not to interface with IrScrutinizer per se, but to allow Broadlink hardware to be used from Home Assistant (or anything else that can talk HTTP and/or MQTT). Home Assistant's built-in integration only supports "Broadlink syntax", while I wanted to be able to use Pronto CCF since that is much closer to a "standard" and there's plenty of codes available. The bridge is just hooking network connectivity on top of a Pronto-to-Broadlink converter.

But I also implemented the LIRC/CCF protocol as a nice extra because I use IrScrutinizer and it was a very convenient way to be able to transmit codes (in order to test them) from within the program. I guess that now I will need to implements the Girs protocol as a replacement.

But if I can vote for what would make my life easier, it would be if IrScrutinizer understood codes in Broadlink syntax, so I could capture them elsewhere and simply paste them in the program and be able to analyze those.

bengtmartensson, 2022-12-31:
Sorry for the delay.

Thank you for telling be the rationale behind your work. 

> But if I can vote for what would make my life easier, it would be if IrScrutinizer understood codes in Broadlink syntax, so I could capture them elsewhere and simply paste them in the program and be able to analyze those.

This has been on my todo list for a long time, cf. #375. I just implemented it, available as the snapshot from this morning; you can just paste Broadlink signals (both hex (the ones starting with 26), and base64 signals. I would be very happy for your testing and feedback. 

> ... if Girs allows Pronto CCF syntax?

"Girs" is not a particularly well defined "standard". However, the description as per the link above explicitly mentions it, and it is implemented [here](https://github.com/bengtmartensson/AGirs) for Arduino.

> I guess that now I will need to implement the Girs protocol as a replacement

Of course, you decide for yourself what you "need" to do. I think it would be quite cool to have Girs support; that way a client could talk both to your server, and for example an (ethernet connected) Arduino using the same driver.

I am not absolutely against reinstating the Lirc-CCF, but the problem is that there is no real plugin/extension architecture, as for the export formats. Lets see...

I also plan to support the Broadlink natively, but I guess I already said that.

bengtmartensson, 2023-01-15:
The parsing of Broadlink format IR signal was checked in some time ago, but there were some problems, which I think are fixed in the commit from today. It also implement the Broadlink export in a better way, as extra formats embedded in  Girs or text exports (#472).

The native support of Broadlink is getting  closer...

lbschenkel, 2023-01-17:
I'm a bit out of free time at the moment to dedicate to this particular project, but I intent do (1) try out your new functionality and (2) implement Girr protocol in my bridge unless the new functionality in IrScrutinizer makes that completely irrelevant — but for that to be the case, it shall be possible to press the "transmit" button in IrScrutinizer and have the code being transmitted by the Broadlink hardware, without any other extra steps. This is the itch that I'm trying to scratch.

commit 8c476725 (2021-07-04): Remove Lirc as sending device. Resolves #399.