bengtmartensson/IrScrutinizer #404: Kubuntu installation AppImage problem

morusa, 2020-11-04:
hi,

"appimage" is supposed to simplify distribution, but it doesn't work on kubuntu :(

$ sudo ./IrScrutinizer-2.3.0-x86_64.appimage 
[sudo] password for user: ***
/tmp/.mount_IrScrumqtoPi/AppRun: 31: exec: IrScrutinizer-2.3.0-x86_64.appimage: not found

bengtmartensson, 2020-11-04:
chmod +x IrScrutinizer-2.3.0-x86_64.appimage

and do not use sudo.

morusa, 2020-11-04:
thanks for answering. I already tried on another computer and the same:

$ chmod +x IrScrutinizer-2.3.0-x86_64.appimage 
$ ./IrScrutinizer-2.3.0-x86_64.appimage 
/tmp/.mount_IrScruVUmedI/AppRun: 31: exec: IrScrutinizer-2.3.0-x86_64.appimage: not found

bengtmartensson, 2020-11-05:
What kubunto version are you using? Both "this" and the "other computer"?

zarpli, 2020-11-05:
> hi,
> 
> "appimage" is supposed to simplify distribution, but it doesn't work on kubuntu :(

please update kubuntu to latest

morusa, 2020-11-05:
I apologize, everything works fine in kubuntu 20.10

bengtmartensson, 2020-11-05:
Ok, nice that it is working. Hope you will like IrScrutinizer.

But I would very much like to know if you have one or more kubunto versions (current or not) which will not run the AppImage.

morusa, 2020-11-07:
IrScrutinizer is a powerful tool, thanks for the linux support!

I guarantee you the AppImage doesn't work on kubuntu 19.10, but on the latest one (19.20) it works great.

morusa, 2020-11-07:
IrScrutinizer is a powerful tool, thanks for the linux support!

I guarantee you the AppImage doesn't work on kubuntu 19.10, but on the latest one (20.10) it works great.

closed 2020-11-07 by morusa