bengtmartensson/IrScrutinizer #406: Troubleshooting Capture

diyjeff, 2020-11-05:
Great program.  So far everything works (that I have tried) except for capture.  Running under windows 10; hardware is GC-100-06.  IrScrutinizer finds my GC-100 (by the way sending works to the same GC-100) but when I try to capture nothing comes over.  I have tried under Scrutinize signal and Scrutinize remote.  Under Capturing hw the test button immediately returns error (No signal received).  Under Scrutinize signal, the Capture button comes back with the same error.  Capture continuous does not appear to get anything.  
I just need a little help in troubleshooting.  As an aside, the Global Cache utility does work.

bengtmartensson, 2020-11-05:
Sorry, but the GC-100-* does not have a learner, so it cannot work. (IIRC, there is (was?) an add-on, I think called GC Learner, that plugged into the serial port, but I have never tried it, It is quite expensive.)

diyjeff, 2020-11-07:
I have the learner.  If you ever decide to add support for the GC-IRL I would be glad to test.  I have written some code to capture the data (it comes over as Global Cache Compressed format) and it looks straight forward.  I have done programming for decades, but not in Java.

bengtmartensson, 2020-11-08:
To support the learner, create a class, say `org.harctoolbox.harchardware.ir.GCLearner`, that `implements IHarcHardware, ICapture` in the support project https://github.com/bengtmartensson/HarcHardware. 

[API doc for GC-IRL](https://www.globalcache.com/files/docs/API-GC-IRL.pdf)

Ok to close?

closed 2020-11-08 by diyjeff