bengtmartensson/IrScrutinizer #407: Scrutinize Signal will not accept any data with length of zero

hallresearch, 2020-11-25:
In previous versions (V1.4.1), if the IR timing contained any zero length times, the program would prompt the user and replace them with the value the user entered (like 0001).

The latest version (V2.3.0) does not seem to do that anymore and requires the trailing 'zeros' to be removed before the signal can be 'Scrutinized'.

Below is a sample of a Apple TV capture
+9180 -4295 +706 -396 +684 -1519 +690 -1511 +683 -1519 +683 -419 +682 -1519 +682 -1520 +682 -1539 +663 -1519 +682 -1520 +682 -1539 +662 -439 +662 -420 +682 -420 +681 -439 +662 -1521 +680 -439 +662 -1519 +681 -1539 +662 -1539 +662 -1519 +682 -439 +663 -1521 +681 -420 +681 -439 +662 -439 +662 -1539 +662 -1520 +681 -1539 +662 -1539 +662 -439 +662 -455 +662 -8665 +9164 -4295 +699 -402 +683 -1519 +683 -1519 +682 -1519 +682 -419 +682 -1519 +682 -1539 +662 -1519 +682 -1519 +682 -1539 +663 -1539 +662 -439 +662 -439 +662 -420 +681 -439 +662 -1519 +682 -420 +681 -439 +663 -1519 +681 -439 +662 -439 +662 -422 +679 -419 +682 -439 +663 -440 +663 -439 +663 -1541 +662 -1520 +681 -1520 +681 -1539 +662 -439 +662 -455 +662 -30000 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0 +0 -0

bengtmartensson, 2020-11-25:
For now, turn off Options -> Invoke signal cleaner; it will get you the old behavior.

Not sure if that is fhe final  solutiion though.

From where do you get that sort of wacky signals?

hallresearch, 2020-11-25:
Bingo! You're the man..

I write firmware for several products we make and have found IRScrutinizer
to be very useful.
My application just dumps an entire block of memory with the times
collected back to my GUI and I paste that into IRScrutinizer.

GREAT JOB on your program.
Previously I used to paste it all into an Excel spreadsheet to get the
final hex values.

There are many customers in the AV industry that want to control everything
via IR (and/or just one remote) and I found your application was the only
one that actually DECODED accurately.

[image: image.png]

bengtmartensson, 2020-11-29:
Thank you for those nice words!

I am checking in a fix that makes the program try without cleaner (and repeat finder) if it fails with them.

commit 34d90ff0 (2020-11-29): interpretString: Robustness fix.

If it it fails with repeat finder or cleaner, try without.
Resolves #407.
closed 2020-11-29 by bengtmartensson