bengtmartensson/IrScrutinizer #408: Weird import/export issue with this nec1 lirc files

irplus-remote, 2020-12-03:
Hi, hope everything is alright, been a long time i did not nag , but i got a small issue now:

The file below imports correctly (there seem to be 64 bits in total not 32 like a regular NEC controller)
[image]

After export to irplus (PRONTO) however only 32 bits are exported. Any idea what to do? 

PRONTO:
0000 006C 0022 0002 015B 00AD 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0016 0016 0041 0016 0016 0016 0041 0016 0041 0016 0041 0016 0016 0016 0041 0016 05F7 015B 0057 0016 0E6C

```xml

# Please take the time to finish this file as described in
# https://sourceforge.net/p/lirc-remotes/wiki/Checklist/
# and make it available to others by sending it to
# <lirc@bartelmus.de>
#
# This config file was automatically generated
# using lirc-0.10.1(default) on Thu Dec  3 15:44:38 2020
# Command line used: -n -d /dev/lirc1
# Kernel version (uname -r): 5.4.72-v7l+
#
# Remote name (as of config file): xmas1
# Brand of remote device, the thing you hold in your hand:
# Remote device model nr:
# Remote device info url:
# Does remote device has a bundled capture device e. g., a
#     usb dongle? :
# For bundled USB devices: usb vendor id, product id
#     and device string (use dmesg or lsusb):
# Type of device controlled
#     (TV, VCR, Audio, DVD, Satellite, Cable, HTPC, ...) :
# Device(s) controlled by this remote:

begin remote

  name  xmas1
  bits           32
  flags SPACE_ENC
  eps            30
  aeps          100

  header       9225  4414
  one           657  1581
  zero          657   505
  ptrail        658
  repeat       9217  2155
  gap          39312
  repeat_gap   95714
  toggle_bit_mask 0x0
  frequency    38000

      begin codes
          power                    0x00FFA25D 0x5163EE00
          poweroff                 0x00FF22DD 0x5163EE00
          sensor                   0x00FF629D 0x5163EE00
          sensoroff                0x00FF02FD 0x5163EE00
          brightplus               0x00FFE21D 0x5163EE00
          brightdown               0x00FFC23D 0x5163EE00
          imeron                   0x00FFE01F 0x5163EE00
          4h                       0x00FFA857 0x5163EE00
          6h                       0x00FF906F 0x5163EE00
          timeroff                 0x00FF6897 0x5163EE00
          8h                       0x00FF9867 0x5163EE00
          10h                      0x00FFB04F 0x5163EE00
          steady                   0x00FF30CF 0x5163EE00
          wave                     0x00FF18E7 0x5163EE00
          phase                    0x00FF7A85 0x5163EE00
          fadephase                0x00FF10EF 0x5163EE00
          twinkle                  0x00FF38C7 0x5163EE00
          fade                     0x00FF5AA5 0x5163EE00
          twinklbtm                0x00FF42BD 0x5163EE00
          combi                    0x00FF52AD 0x5163EE00
      end codes

end remote

```

bengtmartensson, 2020-12-04:
You have found a Lirc file somehow. Semantically correct as a Lirc file, IrScrutinizer imports it; however, it does not (fully) match any of the known protocols. The first part matches the NEC1 though. So if you are selecting a parametrized export, you get the match, namely NEC1. This is the correct behavior, and not a bug.

If you so desire, you can import as raw, and then export as raw. Personally, I would not trust the Lirc file though...

Lirc is a headless chicken: dead but still running...;-).

irplus-remote, 2020-12-05:
I actually did read it via irrecord ot found somewhere :) The remote really sends 2 codes one after another

closed 2020-12-13 by bengtmartensson