bengtmartensson/IrScrutinizer #413: Export code gives Arduino Compile warnings

KingJimmy27, 2021-01-08:
The following sketch was generated using IrScrutinizer with Export Format = Arduino Raw for NEC1 S=124,D=135,F=145, using the VisualMicro environment and gives the warning:
_expresion must have a constant value.
The value of parameters "lengthintro" cannot be used as a constant._

**If I test it in the Arduino environment I dont get the error warning.** but I believe VisualMicro uses the Arduino to generate code.

```
// This Arduino sketch was automatically generated by IrScrutinizer.

// It supports:
//
// * IRremote https://github.com/z3t0/Arduino-IRremote
//            http://z3t0.github.io/Arduino-IRremote/
// * IRLib    https://github.com/cyborg5/IRLib
// * Infrared4Arduino https://github.com/bengtmartensson/Infrared4Arduino,
//                    http://www.harctoolbox.org/Infrared4Arduino.html

// For problems, bugs, and suggestions, please open an issue at
// https://github.com/bengtmartensson/IrScrutinizer/issues

// This file contains C identifiers which have been translated from command names.
// Under some circumstances, these may clash with predefined variables.

// Define exactly one of these
// #define IRREMOTE
// #define IRLIB
#define INFRARED4ARDUINO

/////////////////////////////////////////////////////
// Sanity check
#if defined(IRREMOTE) + defined(IRLIB) + defined(INFRARED4ARDUINO) != 1
#error Must define exactly only one of IRREMOTE, IRLIB, or INFRARED4ARDUINO
#endif

#ifdef IRREMOTE
#include <IRremote.h>
#elif defined(IRLIB)
#include <IRLib.h>
#define sendRaw send
#elif defined(INFRARED4ARDUINO)
#include <IrSenderPwm.h>
#else
#error Must defined one of IRREMOTE, IRLIB, or INFRARED4ARDUINO
#endif

#if defined(IRREMOTE) | defined(IRLIB)
typedef uint16_t microseconds_t; // Change if desired
typedef uint16_t frequency_t;    // Change if desired

static inline unsigned hz2khz(frequency_t f) { return f/1000U; }

// IRremote and IRLib does not have HAS_FLASH_READ or similar.
// As a first guess, use defined(ARDUINO_ARCH_AVR).
// Change if desired.
#define HAS_FLASH_READ defined(ARDUINO_ARCH_AVR)
#endif

// Infrared4Arduino already defines HAS_FLASH_READ, architecture dependant.
// Feel free to override if you think that is a good idea.

#if ! HAS_FLASH_READ
#define PROGMEM /* nothing */
#endif

// Constants
static const long BAUD = 115200UL; // Change if desired

// Global variables
#ifdef IRREMOTE
IRsend irsend;
#elif defined(IRLIB)
IRsendRaw irsend;
#else
IrSender *irsend = IrSenderPwm::getInstance(true);
#endif

// Command #1: IrScrutinizer scrutinized signal
// Protocol: nec1, Parameters: D=135 F=145 S=124
const microseconds_t intro_IrScrutinizer_scrutinized_signal[] PROGMEM = { 9024U, 4512U, 564U, 1692U, 564U, 1692U, 564U, 1692U, 564U, 564U, 564U, 564U, 564U, 564U, 564U, 564U, 564U, 1692U, 564U, 564U, 564U, 564U, 564U, 1692U, 564U, 1692U, 564U, 1692U, 564U, 1692U, 564U, 1692U, 564U, 564U, 564U, 1692U, 564U, 564U, 564U, 564U, 564U, 564U, 564U, 1692U, 564U, 564U, 564U, 564U, 564U, 1692U, 564U, 564U, 564U, 1692U, 564U, 1692U, 564U, 1692U, 564U, 564U, 564U, 1692U, 564U, 1692U, 564U, 564U, 564U, 38628 };
const microseconds_t repeat_IrScrutinizer_scrutinized_signal[] PROGMEM = { 9024U, 2256U, 564U, 65535U };

static void sendRaw
#if HAS_FLASH_READ
(const microseconds_t intro_P[], size_t lengthIntro, const microseconds_t repeat_P[],
             size_t lengthRepeat, frequency_t frequency, unsigned times) {
    microseconds_t intro[lengthIntro];
    microseconds_t repeat[lengthRepeat];
    memcpy_PF(intro, (uint_farptr_t) intro_P, sizeof(microseconds_t) * lengthIntro);
    memcpy_PF(repeat, (uint_farptr_t) repeat_P, sizeof(microseconds_t) * lengthRepeat);
#else // ! HAS_FLASH_READ
(const microseconds_t intro[], size_t lengthIntro, const microseconds_t repeat[],
             size_t lengthRepeat, frequency_t frequency, unsigned times) {
#endif // ! HAS_FLASH_READ

#if defined(IRREMOTE) | defined(IRLIB)
    if (lengthIntro > 0U)
        irsend.sendRaw(intro, lengthIntro, hz2khz(frequency));
    if (lengthRepeat > 0U)
        for (unsigned i = 0U; i < times - (lengthIntro > 0U); i++)
            irsend.sendRaw(repeat, lengthRepeat, hz2khz(frequency));
#else // INFRARED4ARDUINO
    IrSignal irSignal(intro, lengthIntro, repeat, lengthRepeat, NULL, 0U, frequency);
    irsend->sendIrSignal(irSignal, times);
#endif
}

void setup() {
    Serial.begin(BAUD);
    Serial.setTimeout(60000UL);
}

// A pretty silly main loop; just intended as an example.
void loop() {
    Serial.println(F("Enter number of signal to send (1 .. 1)"));
    long commandno = Serial.parseInt();
    Serial.println(F("Enter number of times to send it"));
    long times = Serial.parseInt();
    switch (commandno) {
    case 1U:
        sendRaw(intro_IrScrutinizer_scrutinized_signal, 68U, repeat_IrScrutinizer_scrutinized_signal, 4U, 38400U, times);
        break;
    default:
        Serial.println(F("Invalid number entered, try again"));
        break;
    }
}
```

KingJimmy27, 2021-01-08:
Also the following sketch was generated using IrScrutinizer with Export Format = Arduino/Infrared4Arduino for NEC1 S=124,D=135,F=145 in the Arduino environment 1.8.13 and produces the following warning in **_static void sendRaw_** even though it is not called in the setch:
_1234.ino:32:13: warning: 'void sendRaw(const microseconds_t*, size_t, const microseconds_t*, size_t, frequency_t, unsigned int)' defined but not used [-Wunused-function] static void sendRaw(const microseconds_t intro_P[], size_t lengthIntro, const microseconds_t repeat_P[],  ^~~~~~~_

```
// This Arduino sketch was automatically generated by IrScrutinizer.

// It supports:
//
// * Infrared4Arduino https://github.com/bengtmartensson/Infrared4Arduino,
//                    http://www.harctoolbox.org/Infrared4Arduino.html

// For problems, bugs, and suggestions, please open an issue at
// https://github.com/bengtmartensson/IrScrutinizer/issues

// This file contains C identifiers which have been translated from command names.

#include <IrSenderPwm.h>
#include <Nec1Renderer.h>

// Constants
static const long BAUD = 115200UL; // Change if desired

IrSender *irsend = IrSenderPwm::getInstance(true);

// Command #1: NEC1_D135S124F145
// Protocol: nec1, Parameters: S=124U D=135U F=145U

static void sendNec1(unsigned D, unsigned S, unsigned F, unsigned times) {
    const IrSignal *irSignal = Nec1Renderer::newIrSignal(D, S, F);
    irsend->sendIrSignal(*irSignal, times);
    delete irSignal;
}

static void sendRaw(const microseconds_t intro_P[], size_t lengthIntro, const microseconds_t repeat_P[],
             size_t lengthRepeat, frequency_t frequency, unsigned times) {
    microseconds_t intro[lengthIntro];
    microseconds_t repeat[lengthRepeat];
    memcpy_PF(intro, (uint_farptr_t) intro_P, sizeof(microseconds_t) * lengthIntro);
    memcpy_PF(repeat, (uint_farptr_t) repeat_P, sizeof(microseconds_t) * lengthRepeat);

    IrSignal irSignal(intro, lengthIntro, repeat, lengthRepeat, NULL, 0U, frequency);
    irsend->sendIrSignal(irSignal, times);
}

void setup() {
    Serial.begin(BAUD);
    Serial.setTimeout(60000UL);
}

// A pretty silly main loop; feel free to replace it by something smarter.
void loop() {
    Serial.println(F("Enter number of signal to send (1 .. 1)"));
    long commandno = Serial.parseInt();
    Serial.println(F("Enter number of times to send it"));
    long times = Serial.parseInt();
    switch (commandno) {
    case 1L:
        sendNec1(135U, 124U, 145U, times);
        break;
    default:
        Serial.println(F("Invalid number entered, try again"));
        break;
    }
}

```

bengtmartensson, 2021-01-08:
This is ["variable length array"](https://stackoverflow.com/questions/7812566/why-no-variable-size-array-in-stack), and is not a part of the C++ standard. So it qualifies as a bug. Gcc (which the Arduino environment uses) groks it without messages, however, with either `-Wvla` or `-pedantic`, it warns. @KingJimmy27's compiler warns, but seems to accept it. For this reason, I do not consider a quick (and dirty?) fix necessary.

I intend to fix later, after updating my knowledge of "modern C++".

The second comment was moved to its own issue, #414, and has been fixed.

commit 9559b5ee (2023-02-03): Replace export Arduino raw by Infrared4Arduino raw and IRremote raw,...

together with considerable fixes. Resolves #476. Fixes #413. Also #444 addressed.
closed 2023-02-03 by bengtmartensson