bengtmartensson/IrScrutinizer #419: RCMM32 Protocol

cyber-junkie9, 2021-03-07:
hi there , thank you for such awesome software
can you please help me to decode this IR codes ? i can't find RCMM32 protocol anywhere

https://gist.github.com/ashtam55/95aec3c1072488d345f9b7684831fa6c#airtel-dth---ir-codes-hex

bengtmartensson, 2021-03-07:
It _may_ be the protocol called Nokia32.  Is [this](http://www.remotecentral.com/cgi-bin/mboard/crestron/thread.cgi?523) helpful?

cyber-junkie9, 2021-03-07:
yes , so its called Nokia 32
can you help me to covert it to raw code ?
there are 5 parameters for Nokia32
device , sub , function , toggle , X
but the code is "0X2600270C" (eg-power) . how to convert it ?

bengtmartensson, 2021-03-07:
I do not like the statement that the signal  "is"  0X2600270C; that is some description someone has cooked up, the 5 parameter form is likely the one more close to the manufacturer.  The IRP form of Nokia32 is 

    {36k,1p,msb}<6,-10|6,-16|6,-22|6,-28>((15,-10,D:8,S:8,T:1,X:7,F:8,6,^100m)*,T=1-T) [D:0..255,S:0..255,F:0..255,T@:0..1=0,X:0..127]

so likely 8 bits of D concatenated to 8 bits of S concatenated with 1 bit of T concatenated to 7 bits of X concatenated with 8 bits of F should equal 0x2600270C; i.e. D=26, S=00, T=0, X=27, F=0C (all hexadecimal) is the answer.

bengtmartensson, 2021-03-09:
Transferring to IrpTransmogrifier (it is a protocol issue/question), and to a  discussion.