bengtmartensson/IrScrutinizer #438: IrScrutinizer capture does not work with Girs Nano

zpuskaric, 2021-12-01:
Hi,

I am trying to capture my remotes by using Arduino Nano with Girs 1.03 backend and IrScrutinizer. Girs is uploaded to Nano and in Arduino serial console I can capture remote keys as is visible in the screenshot:

[image]

After that I closed Arduino IDE and started IrScrutinizer-2.3.1-x86_64.AppImage on Ubuntu 10.04. My user is member of dialout group and selecting and configuring Girs backend (Sending hw) shows that device is connected and Girs version is 1.0.3.

[image]

My problem is that capture does not work, for example when I press Test button on Capturing hw I get: "No Signal received"
When I pressed testing button yellow led lights and after reading remote key it goes back to red LED.

Is there a way to log serial connection if what I wrote does not have imediate answer?

zpuskaric, 2021-12-01:
Ubuntu is version 20.04 (not 10.04).

bengtmartensson, 2021-12-02:
Hi,

> Is there a way to log serial connection if what I wrote does not have imediate answer?

Select Options (in the top bar) -> verbose, and the serial communication will be logged to the console window. Please do this, perform the whole procedure,  and post the (complete) content here. Both the screenshoots look fine.

zpuskaric, 2021-12-02:
Here is the log:

LocalSerialPortBuffered.readString: received "?GirsLite 1.0.3"
LocalSerialPortBuffered.sendString: Sent '\r'.
LocalSerialPortBuffered.readString: received "OK"
LocalSerialPortBuffered.sendString: Sent 'version\r'.
LocalSerialPortBuffered.readString: received "GirsLite 1.0.3"
LocalSerialPortBuffered.sendString: Sent 'modules\r'.
LocalSerialPortBuffered.readString: received "Base Transmit Capture 
Receive Led Parameters"
LocalSerialPortBuffered.sendString: Sent 'analyze\r'.
LocalSerialPortBuffered.readString: TIMEOUT
LocalSerialPortBuffered.sendString: Sent 'analyze\r'.
LocalSerialPortBuffered.readString: TIMEOUT

bengtmartensson, 2021-12-02:
```
LocalSerialPortBuffered.readString: received "?GirsLite 1.0.3"
LocalSerialPortBuffered.sendString: Sent '\r'.
LocalSerialPortBuffered.readString: received "OK"
LocalSerialPortBuffered.sendString: Sent 'version\r'.
LocalSerialPortBuffered.readString: received "GirsLite 1.0.3"
LocalSerialPortBuffered.sendString: Sent 'modules\r'.
LocalSerialPortBuffered.readString: received "Base Transmit Capture
Receive Led Parameters"
LocalSerialPortBuffered.sendString: Sent 'analyze\r'.
```

At this time, the Arduino starts looking for an IR signal, and the yellow LED goes on. Please have the to-be-analyzed signal sent here.

    LocalSerialPortBuffered.readString: TIMEOUT

For some reason, nothing arrives (or is recognized).

That is all I can say without further information.

zpuskaric, 2021-12-02:
Here are the same commands and replies from Arduino IDE serial console:

Sent: \r
received: OK
Sent: version\r
received: GirsLite 1.0.3
Sent: modules\r
received: Base Transmit Capture Receive Led Parameters
Sent: analyze\r
<yellow LED turns on>
<button on remote pressed>
<red LED turns on>
received: f=36363 +867 -898 +865 -898 +1752 -898 +865 -898 +865 -899 
+865 -1788 +1750 -898 +865 -1786 +1752 -899 +865 -898 +865 -65535 +865 
-901 +865 -898 +1750 -898 +865 -898 +867 -898 +865 -1786 +1750 -899 +865 
-1788 +1750 -898 +865 -898 +865 -65535

bengtmartensson, 2021-12-03:
Please do not fullquote my text. In this context, it just adds noise. Thank you.

> Here are the same commands and replies from Arduino IDE serial console:
> ...
> received: f=36363 +867 -898 +865 -898 +1752 -898 +865 -898 +865 -899
+865 -1788 +1750 -898 +865 -1786 +1752 -899 +865 -898 +865 -65535 +865
-901 +865 -898 +1750 -898 +865 -898 +867 -898 +865 -1786 +1750 -899 +865 
-1788 +1750 -898 +865 -898 +865 -65535

This is perfect. So the hardware works.

I have just tried with Ubunto 20.04.01 and IrScrutinizer 2.3.1 AppImage and I cannot reproduce. Everything work fine. So I do not know how I can help you...??

zpuskaric, 2021-12-04:
Hi

I am trying to debug the problem and for that matter i installed serial 
port monitoring program called interceptty. It attaches to real serial 
port and creates virtual serial port to which a program can be attached 
(in my case real port is /dev/ttyUSB0 and virtual is /dev/ttyUSB3). I 
can connect to port with minicom and everything works OK (I can send 
commands and am getting expected replies).

When I start IrScrutinizer it does see virtual port (/dev/ttyUSB3) but 
when I try to connect to it (<Open> button in Girs Client Sending hw) it 
returne message: "NonExistingPortException: /dev/ttyUSB3"

Is there a way around it? Perhaps some other serial port monitoring 
program that works with IrScrutinizer?

P.S. interceptty command line is:

interceptty -s 'ispeed 115200 ospeed 115200' /dev/ttyUSB0 /dev/ttyUSB3

bengtmartensson, 2021-12-04:
> Is there a way around it? Perhaps some other serial port monitoring
program that works with IrScrutinizer?

With the verbose option, the serial communication is echoed to the console, so there is no real need for a port-sniffer.

Please flash "my" 1.0.3 using e.g. 
 https://github.com/bengtmartensson/AGirs/releases/download/Version-1.0.3/GirsLite-1.0.3-nano-flasher.sh

Please also try the "Generic Binary" as per http://www.harctoolbox.org/IrScrutinizer.html#Generic+Binary

bengtmartensson, 2021-12-18:
Have you made any further progress on this?

Again, you are describing a sort-of "this-cannot-happen" situation (bidirectional serial communication first works (version number and modules), then it suddenly does not work). And I cannot reproduce...

zpuskaric, 2021-12-21:
Hi,

I intend to try it on another computer but have been busy lately. Will
let you know the outcome.