bengtmartensson/IrScrutinizer #439: decoding Espon RLC bug

kdschlosser, 2021-12-07:
Here are the parameters

d=51
s=101
obc=50
f1=3
f2=0
t1=1
t2=2

If I render the code with the above parameters I get the following output
```
Freq=38400Hz[][+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+577,-1154,+577,-577,+1154,-577,+1154,-1154,+577,-577,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-75000][+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+1154,-577,+577,-577,+1154,-577,+1154,-1154,+1154,-1154,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-250000]
```

If I then scrutinize using the above output I get this as the output

```
Epson: {D=51,F1=3,F2=0,OBC=50,S=101,T1=1,T2=2}
```
which is what is expected based on the values I input into the encoder

In a real world application the RLC can be one of the 2 following RLC examples.

```
[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000, +2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000]
```

```
[[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000], [+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000]]
```

if I decode the top RLC I get this for an output

```
Epson: {D=51,F1=3,F2=0,OBC=50,S=101,T1=1,T2=2}, beg=0, end=59, reps=1
```
which is to be expected based on my input values into the encoder

but when I decode the bottom output I get this as the output.

```
Epson: {D=51,OBC=50,S=101,T1=1,T2=2}
```
F1 and F2 are missing.

The signal view is complete and shows the correct signal tho the second half of it is blue instead of red.
The pasted in RLC gets rebuilt and displays the following

```
Freq=38400Hz[+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+577,-1154,+577,-577,+1154,-577,+1154,-1154,+577,-577,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-75000,+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+1154,-577,+577,-577,+1154,-577,+1154,-1154,+1154,-1154,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-250000][][]
```

which is identical to what the first  RLC gets changed into. 

If I copy the newly created RLC when decoding the second example and then paste it back in and scrutinize it the missing F1 and F2 appear.

Kind of odd behavior for it to convert an RLC that didn't decode properly into a form that when decoded will decode properly. It is also not converting the RLC properly either. In both cases the converted RLC should be the same as what the encoder creates but neither are.

top is what the encoder created and the bottom is what the decoder created
```
Freq=38400Hz[][+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+577,-1154,+577,-577,+1154,-577,+1154,-1154,+577,-577,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-75000][+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+1154,-577,+577,-577,+1154,-577,+1154,-1154,+1154,-1154,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-250000]
Freq=38400Hz[+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+577,-1154,+577,-577,+1154,-577,+1154,-1154,+577,-577,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-75000,+2308,-577,+1154,-1154,+1154,-577,+1154,-1154,+1154,-577,+1154,-577,+577,-577,+1154,-577,+1154,-1154,+1154,-1154,+577,-1154,+577,-1154,+577,-577,+577,-1154,+577,-250000][][]
```

I may not be interpreting this properly or there is an issue with the encoder and decoder I am not sure which. There is an issue with the IRP symbols I do know that for sure.

the top is the IRP symbol from the decoder and the bottom is the IRP symbol from the encoder.

```
{38.4k,577,msb}<1,-1|1,-2|2,-1|2,-2>(4,-1,A:26,1,-75m,4,-1,B:26,1,-250m){A=0x3b92c51,B=0x3ba2f51}
{38.4k,577}<2,-1|1,-2|1,-1|2,-2>((4,-1,D:8,T1:2,OBC:6,T2:2,S:8,1,-75m)*,(4,-1,D:8,~F1:2,OBC:6,~F2:2,S:8,1,-250m))
```

those are 2 completely different symbols.

The first problem is in the top symbol you see "msb" in the general spec but in the bottom one no bit order is defined  which would be "lsb" as a default. 

So first question is what is it msb or lsb?

The other thing is in the top symbol it tells us that there is 1 sequence but in the botom it states there are 2
I am sure you can see what the rest of the issues are.

I am going to hand decode the RLC that was created by the encoder using the bottom symbol. This is what I may not do correctly.

If we look at the encoder symbol it has the timing as 577 and has the burst pairs defined as follows.

```
<
 2,-1  =  (1154, -577)
| 1,-2 =  (577 , -1154)
| 1,-1 =  (577, -577)
| 2,-2 =  (1154, -1154)
>
```

and in binary

```
(1154, -577) = 00
(577 , -1154) = 01
(577, -577) = 10
(1154, -1154) = 11
```

Then we have the bit spec which states there are 2 sequences. and the lead in burst pair for both sequences is (2308, -577) and the first sequence has a lead out of (577, -75000) and the second has a lead out of (577, -250000)

it also states that F1 and F2 have it's bits inverted

apply that to the encoded RLC
```
[
+2308,-577,    = lead in
+1154,-1154,  = 11    
+1154,-577,    = 00
+1154,-1154,  = 11
+1154,-577,    = 00
+577,-1154,    = 01
+577,-577,      = 10
+1154,-577,    = 00
+1154,-1154,  = 11
+577,-577,      = 10 
+577,-1154,    = 01
+577,-1154,    = 01
+577,-577,      = 10
+577,-1154,    = 01
+577,-75000   = lead out
]

[
+2308,-577,    = lead in 
+1154,-1154,  = 11    
+1154,-577,    = 00
+1154,-1154,  = 11
+1154,-577,    = 00
+1154,-577,    = 00
+577,-577,      = 10
+1154,-577,    = 00
+1154,-1154,  = 11
+1154,-1154,  = 11
+577,-1154,    = 01
+577,-1154,    = 01
+577,-577,      = 10
+577,-1154,    = 01
+577,-250000 = lead out
]
```
and that makes the binary for each sequence as follows
```
[11001100011000111001011001]    [11001100001000111101011001]

```

the bit spec identifies the following parameters and the number of bits they use

```
D:8,
T1:2,
OBC:6,
T2:2,
S:8,

D:8,
~F1:2,
OBC:6,
~F2:2,
S:8

```

applied to the binary you get 

```
[     11001100    01    100011    10    01011001    ][    11001100    00    100011    11     01011001    ]
            d             t1       obc        t2          s                         d           ~f1       obc      ~f2          s
```

if we use msb bit order we get the following

```
d = 204
t1 = 1
obc = 35
t2 = 2
s = 89
f1 = 3
f2 = 0
```

t1, t2, f1 and f2 are correct the rest are not.

If we use lsb bit order we get the following

```
[     11001100    01    100011    10    01011001    ][    11001100    00    100011    11     01011001    ]
      00110011    10    110001    01    10011010          00110011    00    110001    11     10011010
             d            t1       obc        t2             s                      d           ~f1       obc      ~f2           s
```

```
d = 51
t1 = 2
obc = 49
t2= 1
s = 154
f1 = 3
f2 = 0
```

the device, f1 and f2 are correct but had the f1 and f2 values been either a 1 or a 2 they wouldn't be.

For the sake of trying I also changed each sequence as a whole to lsb. still not correct.

```
[11001100011000111001011001]    [11001100001000111101011001]
[10011010011100011000110011]    [10011010111100010000110011]

[    10011010    01    110001    10    00110011    ][    10011010    11    110001    00    00110011    ]
          154          1         49         2            51                    154          0          49        3         51 
```

As I said I may not be doing this correctly and if I am not point out where I have made a mistake.  Either way there is still a bug in the notation because when no bit orientation is given the default is lsb so the symbol given by the encoder conflicts with the symbol given by the decoder.

bengtmartensson, 2021-12-08:
Hi Kevin,

The Epson protocol is not exactly widely used or very well known. It is to my knowledge first discussed [here](http://www.hifi-remote.com/forums/viewtopic.php?t=17051). From this thread, I wrote the IRP form

    {38.4k,577}<2,-1|1,-2|1,-1|2,-2>((4,-1,D:8,T1:2,OBC:6,T2:2,S:8,1,-75m)*,(4,-1,D:8,~F1:2,OBC:6,~F2:2,S:8,1,-250m)) [D:0..255,S:0..255,OBC:0..63,T1:0..3,T2:0..3,F1:0..3,F2:0..3]

This is quite an unusual protocol: it lacks an into sequence, and has a repeat- and an ending sequence, (where the latter is mandatory)  that are identical, except for the parameters being different. I do not know any other protocol with that property.

The decoding in IrpTransmogrifier (which is used in IrScrutinizer) can try to decode either an _IrSequence_, or an _IrSignal_ (already decomposed in intro-, repeat- and ending sequence). The decoding is pretty robust if given an IrSequence, but, if given an IrSignal, and the decomposition (in intro-, repeat- and ending-sequence) is not compatible with the IRP, the decoding result may not be as expected. So, for example

> In a real world application the RLC can be one of the 2 following RLC examples.
 
> ... 
     [[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000], [+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000]]

The last IrSignal does not fit the scheme at all: The intro is required to be empty, but isn't. The ending is required to be non-empty, but isn't. So the decode is not what expected. (Strictly speaking, it should possibly be rejected altogether...) (If trying all the numbers as one IrSequence, it will decode to the expected parameters.)

I do nor recognize any decoding bugs in your text.

A "scrutinized" signal is independently tried for decoding and analyzing. The former uses a priori knowledge in the form of the IRP database, the latter uses no a priori knowledge. Therefore, the question

> So first question is what is it msb or lsb?

is as meaningless as the classical "is the glass half full or half empty?".

I am not  going to review you attempted hand-decode. However, I would recommend that you try to get acquainted to [invoking the analyzer of irptransmogrifier from the command line](http://www.harctoolbox.org/IrpTransmogrifier.html#analyze). (You already have it installed along with IrScrutinizer.)

kdschlosser, 2021-12-09:
My issue lies with the code being decoded without prior knowledge of what the protocol is as this information may not be known. and using multiple sequences that are separated by a long pause or using a continuous stream that gets fed into the decoder as the burst pairs are detected and continuing to do that until the code is decoded would yield 2 different decodes.

When I decode the encoded RLC using the IRP symbol from the encoder the parameters gotten are not the same as the ones that are given to the encoder. That raises the question of the accuracy of the decoder/encoder. I know that your programs use the IRP notation in order to do the encoding and decoding. So how could the decode be MSB and the encode be LSB and the decoder decodes the parameters correctly for a generated RLC from the encoder?? It shouldn't because the bits would be in an incorrect order. Get what I am saying?

If the encoder used LSB that would make an input value of say 25 which is `11001` in binary end up as `10011`  because LSB is least significant bit first. If the decoder uses MSB then `10011` is used without changing the bit order  and the resulting value is 19 not 25 as it should be.  So how is it that the decoder comes up with the correct parameters? If I was using parameters where the bits would mirror then it would work but not all of the parameters I am using have mirrored bits for the allowed bit depth for a parameter. 51 = 110011 so LSB or MSB would have the same meaning if the parameter only has 6 bits. but 51 on a parameter that has 8 bits is 00110011 and as lsb it is 11001100 which is 204 and not 51.

so even when I use the exact generated signal from the encoder. which shows the IRP as
```
{38.4k,577}<2,-1|1,-2|1,-1|2,-2>((4,-1,D:8,T1:2,OBC:6,T2:2,S:8,1,-75m)*,(4,-1,D:8,~F1:2,OBC:6,~F2:2,S:8,1,-250m))
```

when decoded the IRP in the decoder is displayed as 
```
{38.4k,577,msb}<1,-1|1,-2|2,-1|2,-2>((4,-1,A:26,1,-75m)*,(4,-1,B:26,1,-250m)){A=0x41911a,B=0x42931a}
```

The decoded value is
```
Epson: {D=166,F1=3,F2=0,OBC=36,S=6,T1=1,T2=1}
```

which is impossible. because the encoded RLC is supposed to be done in LSB and the decoder states that it used MSB to do the decoding. 

in MSB  these are the values.
```
D = 116 = 10100110
F1 = 3 = 11
F2 = 0 = 00
OBC = 36 = 100100
S = 6 = 00000110
T1 = 1 = 01
T2 = 1 = 01
```

if the encoder used LSB like it is supposed to the values would be
```
D = 01100101
F1 = 11
F2 = 00
OBC = 001001
S = 01100000
T1 = 10
T2 = 10
```

If the signal is then decoded as MSB the results would be
```
D = 101
F1 = 3
F2 = 0
OBC = 9
S = 96
T1 = 2
T2 = 2
```

and the final comparison as it should be would be
```
166 != 101
3 == 3
0 == 0
36 != 9
6 != 96
1! = 2
1 != 2
```

So how is the encoder encoding in LSB and the decoder decoding in MSB and the parameters match?? They shouldn't but they do. So there is a bug somewhere.

The other question is why is the decoded IRP symbol so different then the encoder IRP symbol when the correct protocol is identified by the decoder???? shouldn't the IRP symbol in the decoder be 
[... 67 lines cut ...]
where as this
```
+2308,-577,+577,-577,+577,-1154,+577,-577,+577,-577,+577,-1154,+1154,-577,+577,-1154,+577,-577,+577,-1154,+577,-577,+577,-1154,+1154,-577,+1154,-577,+577,-75000, +2308,-577,+577,-577,+577,-1154,+577,-577,+577,-577,+1154,-577,+1154,-577,+577,-1154,+577,-577,+1154,-1154,+577,-577,+577,-1154,+1154,-577,+1154,-577,+577,-250000
```
produces this as a decoded value

```
Epson: {D=166,F1=3,F2=0,OBC=36,S=6,T1=1,T2=1}, beg=0, end=59, reps=1
```

Now sure why there is extra data on the end of the decoded value. 

There is also another problem with the IRP symbol. There being a * after the first sequence is wrong and I do not believe the F1 and F2 are user settable parameters. Here is why.

I am not sure why there is a * and not a + after the first sequence because in order to decode the protocol properly the first and last sequence MUST exist  when using the IRP symbol that is given for the protocol. Otherwise the signal cannot be decoded properly. Unless the F1 and F2 are toggle bits and should be the inverse of T1 and T2 for when the button gets released. Otherwise there would be no way to have the first sequence repeat because the code would not be able to be decoded until the ending sequence was received which really wouldn't make sense  because the device receiving the signal would not be able to react until the ending sequence was given??. and the number of repeats would have to be counted and then executed once the button gets released.. Done know if holding the button down for 100 repeats and a device doing nothing until it is released and the device executing the command 100 times after the button has been released would be how it works. 

This kind of a notation would make a whole lot more sense.
```
{38.4k,577}<2,-1|1,-2|1,-1|2,-2>((4,-1,D:8,T1:2,OBC:6,T2:2,S:8,1,-75m)+,(4,-1,D:8,~T1:2,OBC:6,~T2:2,S:8,1,-250m))                [D:0..255,S:0..255,OBC:0..63,T1:0..3,T2:0..3]
```
With how the notation is currently done it would work because if the same code is not received within 108466 microseconds or the code that has been received had incorrect F1 and F2 it would consider the button as being released anyhow.

why there would be 4 values that T1 and T2 could be set to is something that I do not know. but I would imagine that there is a relationship between T1 and F1 and T2 and F2 only for the purposes of indicating that the button has been released.

All I can tell you is that manually decoding a signal that is created by the program does not decode to the same values that were put in when using either the IRP Symbol seen in the decoder or in the encoder. And the  decoder does not handle repeat codes.

kdschlosser, 2021-12-09:
I also used this as an example because a program that is reading the ir signals is either going to feed in a stream of marks and spaces like this

```
+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000, +2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000
```

Or if the program looks for large gaps and separates the received marks and spaces and then feeds in the sequences when the first large gap is seen it would attempt to decode this

```
+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000
```
an when that did not properly decode it would hold onto the last sequence and when the next large gap is seen it would attempt to feed in the last one along with the  new one 
```
[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000]
[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000]
```
which one would think would decode properly. as the program should try to decode it as the following first
```
[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000]
[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000][]
```

and when it was unable to it would attempt to decode it as

```
[][+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000]
[+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +577, -1154, +577, -577, +1154, -577, +1154, -1154, +577, -577, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -75000]
```
which would be successful

But given that the first sequence of the  Epson protocol can repeat a proper decode of this signal should occur
```
+2308, -577, +1154, -1154, +1154, -577, +1154, -1154, +1154, -577, +1154, -577, +577, -577, +1154, -577, +1154, -1154, +1154, -1154, +577, -1154, +577, -1154, +577, -577, +577, -1154, +577, -250000
```

but because F1 and F2 only appear in the sequence after all repeats a complete decode cannot be done until after the button has been released. and when all received marks and spaces are used to identify the signal no repeats are indicated and he code gets truncated  using only the last 2 sequences t be decoded.

kdschlosser, 2021-12-09:
x10.n protocol also has problems with decoding.

```
{40.8k,565}<2,-12|7,-7>(F:5,N:-4,21,-7,(7,-7,F:5,~F:5,21,-7)+,N=(N+1)%16)[F:0..31,N@:0..15=0]
```

```
Freq=40800Hz[+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][]
```

the second sequence shouldn't be there if there are no repeats. And if I paste in 
```
Freq=40800Hz[+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][][]
```

into the decoder it does not decode even tho it is a correct signal. The + after the second sequence in the IRP symbol indicates 1 OR more and the first occurrence of the the second sequence appears in the 1 sequence of the created RLC. So technically speaking the encoder is also not encoding properly because There is no place to specify the number of repetitions so the encoded RLC should be
```
Freq=40800Hz[+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][][]
``` 

You can see in the first sequence of the RLC the first occurrence of the repeat sequence

Freq=40800Hz[+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+11865,-3955,<strong><em>+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955</em></strong>][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][]

and if I paste in the following
```
[+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955]
```

I get the error "Could not parse text in capture data window as IR signal" even tho the above is correct and is a proper signal with 5 repeats.

I also tried this as well
```
[+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+1130,-6780,+1130,-6780,+3955,-3955,+1130,-6780,+3955,-3955,+3955,-3955,+3955,-3955,+1130,-6780,+11865,-3955][]
```

and it does not decode. 

Does this program not handle repeats?? I see that it states "# reps" in the decoder and I thought that to be repetitions (repeats)