bengtmartensson/IrScrutinizer #444: IRremote export: update and extend

bengtmartensson, 2021-12-29:
[IrRemote](https://github.com/Arduino-IRremote/Arduino-IRremote) has recently been changed in an incompatible way. The export format with the same name should be updated for the new API. 

Also, there are several protocols in IRremote that currently are not supported but it would be easy to do so.

lucki1000, 2022-08-10:
Is there some progress on fixing the export to IrRemote?

bengtmartensson, 2022-08-14:
Sorry, no. I am pretty busy for the moment. But it is not that hard, or that much work. Can you help?

lucki1000, 2022-08-16:
Hi, sorry I really don't have Java or object oriented programming skills, maybe If you can give me some advise I can try to bring it back to a working state. :)

bengtmartensson, 2022-08-17:
It is not really that complicated; knowhow of Java or OO are not required. You only change the file `exportformats.d/IRremote.xml` (and, possibly, the file `exportformats.d/Arduino_Raw.xml`). The procedure is described [here](http://www.harctoolbox.org/IrScrutinizer.html#Adding+new+export+formats). Good luck!

commit 68a2c9d8 (2025-02-10): Fixes to IRremote export (several dialects). Resolves #444.
closed 2025-02-10 by bengtmartensson