bengtmartensson/IrScrutinizer #448: AppImage, Mac & Windows w/ embedded Java

bengtmartensson, 2022-02-19:
Opening https based files in the RemoteLocator fails with SSL handshake error.

dtcooper, 2022-07-24:
Still happening, even on latest 2.4 snapshot. Thanks! See attached screenshot on latest commit as of this writing (`062f9565`).

[image]

bengtmartensson, 2022-07-25:
Sorry, I cannot reproduce. I.e., for me it is working. Exactly what platform are you you using? Can you do something to pinpoint the problem? Can it be a firewall issue?

dtcooper, 2022-07-26:
Not using a firewall, using an Apple M1, 2020 Macbook Air, on macOS 12.4.

bengtmartensson, 2022-08-07:
Ok, I can now reproduce on my Mac (ancient MacMini "early 2009", MacOS 10.11.6) with embedded Java, but it still works on all other platforms. So, as far as I understand, what happens is this: It breaks on line 52 in org.harctoolbox.remotelocator.Girrable.java, when opening an https stream from cdn.jsdelivr.net. This probably uses a crypto that the Mac-Java does not know, as opposed to the other Java JVMs. So I am reopening as bug. 

Unfortunately, I do not have capacity right now to go after it right now, but if someone else... A work-around would be to install a modern Java on the Mac, and using the binary distribution.

Thank you for bringing this problem to my attention.

bengtmartensson, 2023-12-28:
I have decided to cease support for the MacOS version (with embedded Java), #512. 

> A permanent solution would be either to use another Java, or to somehow pack the missing cryptos into the existing one.

So with the first option out of the question, patching up the current Java 11 simply do not seem meaningful. Probably it is do-able, and a contributed fix will be welcome.

So I am closing this. The problem in this issue do not occur in other versions. So it will remain a "known issue" for the, from now unsupported, MacOS distro.

commit c8b31af4 (2022-02-20): Added missing java module to embedded Javas. Resolves #448.
closed 2023-12-28 by bengtmartensson