bengtmartensson/IrScrutinizer #455: Export removes Function details and can't import them

duhow, 2022-04-10:
Learning IR codes with IrScrutinizer, then exporting as Girr, returns me the following XML file:
```xml
<?xml version="1.0" encoding="WINDOWS-1252" standalone="no"?>
<!--This file is in the Girr (General IR Remote) format, see http://www.harctoolbox.org/Girr.html-->
<remotes xmlns="http://www.harctoolbox.org/Girr" xmlns:html="http://www.w3.org/1999/xhtml" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" girrVersion="1.2" title="IrScrutinizer parametric export" xsi:schemaLocation="http://www.harctoolbox.org/Girr http://www.harctoolbox.org/schemas/girr_ns-1.2.xsd">
    <adminData>
        <creationData creatingUser="duhow" creationDate="2022-04-09_23:03:49" tool="IrScrutinizer" tool2="IrpTransmogrifier" tool2Version="1.2.10" toolVersion="2.3.1"/>
    </adminData>
    <remote comment="Export from IrScrutinizer" deviceClass="radiator" manufacturer="Orbegozo" model="REH 1060" name="remote">
        <commandSet name="commandSet">
            <parameters protocol="F12_relaxed">
                <parameter name="D" value="3"/>
                <parameter name="S" value="1"/>
            </parameters>
            <command master="prontoHex" name="Set">
                <ccf>0000 006D 0000 000C 0030 0010 0030 0010 0010 0030 0030 0010 0030 0010 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0030 050F</ccf>
            </command>
            <command master="prontoHex" name="Lock">
                <ccf>0000 006D 0000 000C 0030 0010 0030 0010 0010 0030 0030 0010 0030 0010 0010 0030 0010 0030 0010 0030 0030 0010 0010 0030 0010 0030 0010 052F</ccf>
            </command>
            <command master="prontoHex" name="Mode">
                <ccf>0000 006D 0000 000C 0030 0010 0030 0010 0010 0030 0030 0010 0030 0010 0010 0030 0010 0030 0030 0010 0010 0030 0010 0030 0010 0030 0010 052F</ccf>
            </command>
            <command master="prontoHex" name="Heat">
                <ccf>0000 006D 0000 000C 0030 0010 0030 0010 0010 0030 0030 0010 0030 0010 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0030 0010 0010 052F</ccf>
            </command>
            <command master="prontoHex" name="Up">
                <ccf>0000 006D 0000 000C 0030 0010 0030 0010 0010 0030 0030 0010 0030 0010 0010 0030 0010 0030 0010 0030 0010 0030 0030 0010 0010 0030 0010 052F</ccf>
            </command>
            <command master="prontoHex" name="Down">
                <ccf>0000 006D 0000 000C 0030 0010 0030 0010 0010 0030 0030 0010 0030 0010 0010 0030 0030 0010 0010 0030 0010 0030 0010 0030 0010 0030 0010 052F</ccf>
            </command>
        </commandSet>
    </remote>
</remotes>
```

Parameter F was returned at the moment of learning the code, however in the export is missing.
After closing the program and **importing** this same file, IrScrutinizer will complain regarding "Name F not assigned".
Using "Add missing Fs" adds all possibilities, except filling the missing ones.

### Expected behaviour

Export `parameter name="F"` in each `command` entry.

### Affected version

2.3.1 , also tested on 2.4.0-snapshot e73afa0aab13dd00bbe3cd428d43777fda2d6c2b

duhow, 2022-04-10:
Yikes, I think I sort of figured out, but still the process is not clear...

[image]

I am exporting via the "Scrutinize remote > Export" button, but there is another Tab "Export" that has **Parameters** disabled... by default.

bengtmartensson, 2022-04-10:
This is an [already known problem](https://github.com/bengtmartensson/Girr/issues/30). Although I agree that it looks confusing, it is strictly speaking not a bug: You are requesting an export _without_ the parameters. The generated file can be imported as raw, but not as parametric, which is the expected behavior. If you check "Parameters" on the export pane, it will contain useful parametric representation.

BTW, I discovered a [new problem](https://github.com/bengtmartensson/Girr/issues/33).

Ok to close as duplicate?

duhow, 2022-04-10:
Sounds ok. Thanks!

closed 2022-04-10 by duhow