bengtmartensson/IrScrutinizer #464: harchardware error when invoked from AppImage

JanCeuleers, 2022-12-05:
I have installed IrScrutinizer-2.4.0-SNAPSHOT-x86_64.AppImage in /usr/local/bin and have created two symlinks as follows:

root@bajor:/usr/local/bin# ln -s IrScrutinizer-2.4.0-SNAPSHOT-x86_64.AppImage irscrutinizer
root@bajor:/usr/local/bin# ln -s IrScrutinizer-2.4.0-SNAPSHOT-x86_64.AppImage harchardware
root@bajor:/usr/local/bin# chmod a+x IrScrutinizer-2.4.0-SNAPSHOT-x86_64.AppImage

IrScrutinizer starts OK when run as a normal user. However, harchardware does not:

janc@bajor:~$ harchardware help
/tmp/.mount_harchaQt3NDE/AppRun: 18: basename: Argument list too long
/tmp/.mount_harchaQt3NDE/AppRun: 31: exec: : Permission denied

JanCeuleers, 2022-12-05:
Furthermore, /tmp is full of fuse mount entries:

root@bajor:~# ls -a /tmp/.mount*
ls: cannot access '/tmp/.mount_harcha007kWB': Permission denied
ls: cannot access '/tmp/.mount_harcha0DheLA': Permission denied
ls: cannot access '/tmp/.mount_harcha0iQ85A': Permission denied
ls: cannot access '/tmp/.mount_harcha0MGgRC': Permission denied
ls: cannot access '/tmp/.mount_harcha0NaesB': Permission denied
ls: cannot access '/tmp/.mount_harcha0nc1fz': Permission denied
ls: cannot access '/tmp/.mount_harcha0zZbKA': Permission denied
ls: cannot access '/tmp/.mount_harcha1lyO6z': Permission denied
ls: cannot access '/tmp/.mount_harcha2dgq3A': Permission denied
ls: cannot access '/tmp/.mount_harcha2kuXHy': Permission denied
ls: cannot access '/tmp/.mount_harcha2trYGB': Permission denied
ls: cannot access '/tmp/.mount_harcha2u75LA': Permission denied
ls: cannot access '/tmp/.mount_harcha36hT6B': Permission denied
(...)

bengtmartensson, 2022-12-05:
On the [Lirc mailing list](https://sourceforge.net/p/lirc/mailman/message/37744184/)  I made the claim that if the AppImage is called with a basename of `harchardware`, the main routine of [HarcHardware](https://github.com/bengtmartensson/HarcHardware) is invoked instead of IrScrutinizer. This statement is not true, and I apologize for it. What is true however, is that if the AppImage is invoked with the basename of `irptransmogrifier` the [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier) main routine will be invoked. 

Jan, you will have to install IrScrutinizer using the [generic distribution](http://www.harctoolbox.org/IrScrutinizer.html#Generic+Binary); sorry for the confusion. Accordingly, I will close this issue.

Nevertheless, currently if the AppImage is called with a basename different from `irscrutinizer` and `irptransmogrifier`, something in the path, possibly not in the AppImage (like /usr/local/lib/harctoolbox) may be invoked . This is confusing and qualifies as a bug, and will be fixed in another commit.

closed 2022-12-07 by bengtmartensson

bengtmartensson, 2023-12-17:
By 4f38a129e2b6f0b1ae7311a7a088b6e2dfb02196, harchardware invocation is now supported.