bengtmartensson/IrScrutinizer #470: Syntax error in setup script

JanCeuleers, 2022-12-24:
root@bajor:/usr/local/share/irscrutinizer# ./setup-irscrutinizer.sh 
./setup-irscrutinizer.sh: 13: [: !=: unexpected operator
./setup-irscrutinizer.sh: 13: [: !=: unexpected operator
./setup-irscrutinizer.sh: 13: [: !=: unexpected operator
./setup-irscrutinizer.sh: 13: [: !=: unexpected operator
./setup-irscrutinizer.sh: 13: [: !=: unexpected operator
./setup-irscrutinizer.sh: 13: [: !=: unexpected operator
Creating /usr/local/share/applications/irscrutinizer.desktop succeded.
Creating /usr/local/share/applications/girr.xml succeded.
Consider deleting old properties with the command 
    irscrutinizer --nuke-properties

The above is with dash as the default shell, but I have also tried running the script with various alternative shells, including bash, tcsh and ksh, and they all fail.

This is with setup-irscrutinizer.sh from the 2.4.0 binary zipped distribution.

bengtmartensson, 2022-12-24:
Hi Jan, and a merry X-mas!

I am sorry, I cannot reproduce (tried Fedora and Ubunto). What OS are you using? 

First try to change `/bin/sh` to `/bin/bash`  in line 1; does it change anything? Also try both as root and non-root. Your default shell should not have any effect.

JanCeuleers, 2022-12-24:
I should have added that my unsuccessful attempts have been as follows:

# history | grep setup-irscrutinizer
 2038  ./setup-irscrutinizer.sh
 2040  sh setup-irscrutinizer.sh
 2051  ksh setup-irscrutinizer.sh
 2052  dash setup-irscrutinizer.sh
 2053  csh setup-irscrutinizer.sh
 2056  tcsh setup-irscrutinizer.sh
 2057  zsh setup-irscrutinizer.sh

Weird huh?

bengtmartensson, 2022-12-24:
What is weird is that you ignore my suggestions and questions, and embark on a priori meaningless tests instead. Makes it hard to help you.

JanCeuleers, 2022-12-25:
Sorry Bengt, that's because you missed my previous email (which I had
sent to the wrong email address such that it doesn't appear in the ticket).

Here's what I sent earlier:

===begin===

Hi Bengt,

Merry X-mas to you too.

My OS is Ubuntu 20.04. My attempts have all been as root (this is an
installation script after all).

I can now no longer reproduce either on the machine on which this
happened because the installation has succeeded (I commented out the if
on line 13 and the corresponding fi a few lines down).

Let me try on another machine.

On another machine, also running Ubuntu 20.04, also as root, the
installation succeeded without errors. Very strange.

Better close this.

(Will be asking you a question on the Lirc list in a minute).

Jan

===end===

So my later follow-up, showing the history of commands, was intended as
additional information showing how I had tried running the script using
multiple different shells.

In my understanding, when a script is invoked by including it on the
command line of a specific shell, the shebang is ignored. So I believe
that these invocations were equivalent to changing the shebang in the
script in the way you suggested.

bengtmartensson, 2022-12-25:
> Sorry Bengt, that's because you missed my previous email (which I had
> sent to the wrong email address such that it doesn't appear in the ticket).

FWIW, I have not received any emails pertaining this issue.

The script is written in the bash language, and will fail in "classic" Bourne shell sh. However, many systems link sh to bash, on my Fedora 37:

    $ ls -l /bin/sh
    lrwxrwxrwx. 1 root root 4 Nov 18 15:27 /bin/sh -> bash*

However, it is not a good idea to assume that this is the case. (Also, I have hear somewhere that bash disables some of its cleverness if invoked by root, therefore the question.) So I changed the shebang in a commit listed above. Give what you have said, I think this should be considered the solution, and we can close the issue.

Still, I think I have seen that error message somewhere before, and I would like to understand what causes it...

closed 2022-12-25 by bengtmartensson