bengtmartensson/IrScrutinizer #477: Fix export format Homeass-Broadlink IR

bengtmartensson, 2023-02-03:
Said export format is broken and to be fixed.

In the future extend for 433 MHz RF.

GMichu, 2023-02-12:
How is this export format broken? I found this wonderful project while looking for a way to convert various formats into homeass-broadlink. So far, I have not encountered problems, so I ask to know what to look out for.

bengtmartensson, 2023-02-12:
First, the lenght parameter in the data is wrong: just counts the number of unencoded durations instead of the number of bytes after the 1--2 byte coding. Secondly, the command names were wrong, I think they were all called  power_on or power_off or so. 

I just checked in a partial fix; fixes the second problem but not the first one. (You can just drop the file https://raw.githubusercontent.com/bengtmartensson/IrScrutinizer/master/src/main/config/exportformats.d/Homeass_broadlink.xml into your exportformats.d directory, and you will be set.)  It is possible that the reading programs or hardware does not check the lenght. Please give feedback.

commit cd68b827 (2023-09-01): Two new/fixed export formats:

Homeassistant Broadlink Switch and Homeassistant Broadlink remote. Resolves #477.
closed 2023-09-01 by bengtmartensson