bengtmartensson/IrScrutinizer #480: Support OpenBeken export format

OctoNezd, 2023-02-08:
[OpenBeken](https://github.com/openshwprojects/OpenBK7231T_App) is firmware replacement for cheap Tuya devices running custom tuya-made chips. The syntax of IRSend seems limited to me, not sure whether it can be implemented?

bengtmartensson, 2023-02-11:
Sounds interesting, but please tell me exactly what such an export format is, and what it should do. Reference? Tuya is mostly homeautomation in general, what is a Tuya IR device? 

There are several programs called IRSend (for example in Lirc). Is that a program within Openbeken? 

Do you know [Lirconidan](https://github.com/bengtmartensson/Lirconian)? This is a pure-python implementation of Lirc irsend. Can that be used to re-engineer Openbeken IRSend?

OctoNezd, 2023-02-12:
Tuya released a bunch of IR remotes that are programmable to custom actions, e.g. one can be used for controlling your lighting, AC and whatever you have from your phone and home automation system. With custom firmware they can transmit any IR data, I assume. 

Example of such remote:
[image]

[IRSend is a command in OpenBeken firmware](https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands-extended.md), with what seems a very limited syntax:
<html><body>

IRSend | [PROT-ADDR-CMD-REP] | Sends IR commands in the form PROT-ADDR-CMD-REP, e.g. NEC-1-1A-0 | File: driver/drv_ir.cppFunction: IR_Send_Cmd
-- | -- | -- | --

</body>
</html>

As one of possibilities I hadn't tested yet it may be possible to just change pin configuration from IRSend to 0/1 and manually send a script file to scripting engine port: https://github.com/openshwprojects/OpenBK7231T_App#scripting-engine which may end up in configuration like:
```
setChannel PIN_NUMBER_HERE 1
delay_ms 9
setChannel PIN_NUMBER_HERE 0
delay_ms 0
```

OctoNezd, 2023-02-12:
Just tested - setting IR to Rel and restarting device does make it just turn IR on and off when you click it in webui - gonna try making simple script based on export data of delays from some of IRScrutinizers exports, but I guess the ability to somehow understand what protocols IRSend actually supports and export in that format is better.

closed 2023-02-16 by bengtmartensson