bengtmartensson/IrScrutinizer #482: Export irToy text with LIRC import not working

HeadHodge, 2023-02-27:
Using import file: 
`
#
# this config file was automatically generated
# using lirc-0.8.5-CVS(default) on Thu Apr  9 18:12:26 2009
#
# contributed by
#
# brand:                       Vizio remote
# model no. of remote control: 0980-0305-3000
# devices being controlled by this remote:
#

begin remote

  name  Vizio
  bits           16
  flags SPACE_ENC
  eps            30
  aeps          100

  header       8966  4400
  one           567  1596
  zero          567   494
  ptrail        575
  repeat       9000  2194
  pre_data_bits   16
  pre_data       0x20DF
  gap          40403
  repeat_gap   95450
  toggle_bit_mask 0x0

      begin codes
          KEY_POWER                0x10EF                    #  Was: POWER
      end codes

end remote
`
Goto to scrutinize signal, then click export with export set to: irToy. the console says success but the resulting text is created but empty.

bengtmartensson, 2023-02-27:
Thank you for this report. Also the export format IrToy-bin is currently broken. (It has nothing to do with the Lirc imports.)

closed 2023-02-27 by bengtmartensson

HeadHodge, 2023-02-27:
where is the source code file for the irToy? i looked on github but couldnt find it

bengtmartensson, 2023-02-27:
The official repository is [here](https://github.com/DangerousPrototypes/USB_IR_Toy), but it appears to be lacking the latest version. You can ask in [its offical forum](http://dangerousprototypes.com/forum/index.php?board=29.0). You can also try to ask the IrDroid people.

bengtmartensson, 2023-02-27:
Huh? The issue was closed as "completed" because I checked in a fix. If that does not work, please tell me so. [Snapshot release](https://github.com/bengtmartensson/IrScrutinizer/releases/tag/ci-build).

HeadHodge, 2023-02-27:
> Huh? The issue was closed as "completed" because I checked in a fix. If that does not work, please tell me so. [Snapshot release](https://github.com/bengtmartensson/IrScrutinizer/releases/tag/ci-build).

Hi, saw you duplicated problem, but missed where you fixed it. Not use to such speedy turn around! 🤓👍 

I'm using Win10 and dont see an 'exe' file in your snapshot. So i dont think i can test without one.

Just curious, if you didn't write your own irToy plugin, are you using an existing irToy dll or something to export?

thanks

HeadHodge, 2023-02-28:
hi,

i'm testing your sending hw for irToy. For testing i'm using my usb cdc serial com port (com2). when i configure the irToy settings to com2 and click on the open button, i receive '0x0,0x0,0x0,0x0,0x0,0x76' from you and in return i send you the version number.

but my send fails and you fail with a receive timeout.

i think this might be because your using hardware handshaking and i'm not. if you use handshaking, is there a way to disable it?

HeadHodge, 2023-03-03:
> but my send fails and you fail with a receive timeout.
> 
> i think this might be because your using hardware handshaking and i'm not. if you use handshaking, is there a way to disable it?

looking more into it, it appears that irDroid ran into the same problem and figured a way around it. Maybe changed usb cdc stack? But I would be happy if there just was a way to turn off handshaking on your app

-thanks

HeadHodge, 2023-03-03:
OR.... It would be great if you could export (text and binary formats) to a standard serial com port with 3 settings *com port, *speed parity, and *handshake

commit 4e034e7e (2023-02-27): Export formats  IrToy and IrToy-bin fixed.

Assumed girr:remote as root element. Fixes #482.