bengtmartensson/IrScrutinizer #490: /dev/lirc does not work

bryder, 2023-08-19:
When I change the device (I have /dev/lirc1 and /dev/lirc0) I get:

```
Exception in thread "AWT-EventQueue-0" org.harctoolbox.ircore.ThisCannotHappenException: Unhandled property: PROP_PORTNAME
	at org.harctoolbox.irscrutinizer.GuiMain.lambda$setupDevLirc$39(GuiMain.java:742)
```

And I Open the port but it doesn't work. 

The Test Capture button is not enabled for my receiving device.

This fixed it for me:
```
diff --git a/src/main/java/org/harctoolbox/irscrutinizer/GuiMain.java b/src/main/java/org/harctoolbox/irscrutinizer/GuiMain.java
index e9ad6084..c59c8ced 100755
--- a/src/main/java/org/harctoolbox/irscrutinizer/GuiMain.java
+++ b/src/main/java/org/harctoolbox/irscrutinizer/GuiMain.java
@@ -730,7 +730,7 @@ public final class GuiMain extends javax.swing.JFrame {
         // devLircBean was constructed and setup in initComponents()
         devLircBean.addPropertyChangeListener((PropertyChangeEvent evt) -> {
             switch (evt.getPropertyName()) {
-                case HardwareBean.PROP_PORT:
+                case HardwareBean.PROP_PORTNAME:
                     properties.setDevLircName((String) evt.getNewValue());
                     break;
                 case HardwareBean.PROP_PROPS:

``` 

I'm happy to make a pull request for this if you like.

Once I made that change I could happily capture

I built a version for Rasperry PI 4B using all of your current repos.
I had to build tonto separately since it's not on maven as far as I could tell so I installed it locally.

bengtmartensson, 2023-08-19:
Hi Bill,
thank you very much for this. Yes, I would be happy for a PR.

Note (mainly to myself): HardwareBean contains both PROP_PORT and PORT_PORTNAME. These are for different usages: The latter is for port names, like /dev/lirc0, the first is for port numbers in the sense of GlobalCache ports or TCP port numbers.

The script for building tonto used by the snapshot build is found in common/scripts/build-tonto.sh, in case you need it.  Building the shared library is somewhat tricky, but for IrScrutinizer, we do not need it.

bryder, 2023-08-20:
Pull request #491 submitted.

Thanks for the pointer to the build-tonto.sh script. I did discover it after I built tonto and figured out how to install it into my local maven repo. It would have saved me some time.

I should have looked in common/scripts before since you have the build-harctoolbox-project.sh script which would have saved me some time fumbling around with maven. 

Thanks for the project!

martin3000, 2023-08-21:
I have IrScrutinizer AppImage 2.4.0
I go to the /dev/lirc tab, select /dev/lirc0 and press open. Then I get:

>Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: 'boolean org.harctoolbox.devslashlirc.Mode2LircDevice.canMeasureCarrier()'
        at org.harctoolbox.devslashlirc.Mode2LircDevice.canMeasureCarrier(Native Method)
	at org.harctoolbox.devslashlirc.Mode2LircDevice.toString(Mode2LircDevice.java:167)
	at org.harctoolbox.harchardware.ir.DevLirc.toString(DevLirc.java:300)
	at org.harctoolbox.guicomponents.DevLircBean.setProps(DevLircBean.java:153)
	at org.harctoolbox.guicomponents.DevLircBean.open(DevLircBean.java:210)
	at org.harctoolbox.guicomponents.HardwareBean.openClose(HardwareBean.java:99)

Is this related to this issue?