bengtmartensson/IrScrutinizer #492: Problem opening /dev/lirc0

martin3000, 2023-08-21:
I have ubuntu 22.04 and IrScrutinizer AppImage 2.4.0
I go to the /dev/lirc tab, select /dev/lirc0 and press open. Then I get:

>Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: 'boolean 
org.harctoolbox.devslashlirc.Mode2LircDevice.canMeasureCarrier()'
        at org.harctoolbox.devslashlirc.Mode2LircDevice.canMeasureCarrier(Native Method)
	at org.harctoolbox.devslashlirc.Mode2LircDevice.toString(Mode2LircDevice.java:167)
	at org.harctoolbox.harchardware.ir.DevLirc.toString(DevLirc.java:300)
	at org.harctoolbox.guicomponents.DevLircBean.setProps(DevLircBean.java:153)
	at org.harctoolbox.guicomponents.DevLircBean.open(DevLircBean.java:210)
	at org.harctoolbox.guicomponents.HardwareBean.openClose(HardwareBean.java:99)

bengtmartensson, 2023-08-21:
Sorry for that. Unfortunately, that version seems to use an erroneous version of the DevSlashLirc jar. It should work in the [current shapshot version](https://github.com/bengtmartensson/IrScrutinizer/releases/tag/ci-build), so please try that one.

martin3000, 2023-08-21:
Thank you, that works!

closed 2023-08-21 by martin3000

bengtmartensson, 2023-08-21:
Thank you for the report. I have tagged it as "not a bug" since it is no source code bug, but was a silly build from my side. (Built with a non-checked in. experimental DevSlashLirc).