bengtmartensson/IrScrutinizer #520: exception on import or lirc config

clach04, 2024-03-09:
I've tried:

  * https://github.com/bengtmartensson/IrScrutinizer/releases/tag/Version-2.4.0
  * https://github.com/bengtmartensson/IrScrutinizer/releases/tag/ci-build (from 2024-02-28)

With java 1.8:

    openjdk version "1.8.0_232"
    OpenJDK Runtime Environment (build 1.8.0_232-09)
    OpenJDK 64-Bit Server VM (build 25.232-b09, mixed mode)

And I'm unable to import an lirc file that I created with `irrecord` from lirc that works with lirc (e.g. `irw` and other clients work great and respond). However I'm new to lirc so it maybe there is something odd about the lirc file I have (which is inline below, GitHub would not allow the file to be attached):

    begin remote
    
      name  sony_dvd_RMT-D197A
      driver devinput
      bits           24
      eps            30
      aeps          100
    
      one             0     0
      zero            0     0
      pre_data_bits   40
      pre_data       0x4000400
      gap          44733
      toggle_bit_mask 0x0
      frequency    38000
    
          begin codes
              KEY_PLAY                 0x1A4932
              KEY_PAUSE                0x1A4939
              KEY_STOP                 0x1A4938
              KEY_NEXT                 0x1A4931
              KEY_PREVIOUS             0x1A4930
              KEY_1                    0x1A4900
              KEY_2                    0x1A4901
              KEY_3                    0x1A4902
              KEY_0                    0x1A4909
              KEY_VOLUMEDOWN           0x010013
              KEY_VOLUMEUP             0x010012
              KEY_4                    0x1A4903
              KEY_5                    0x1A4904
              KEY_6                    0x1A4905
              KEY_7                    0x1A4906
              KEY_8                    0x1A4907
              KEY_9                    0x1A4908
              KEY_POWER                0x1A4915
          end codes
    
    end remote

With the latest stable 2.4.0 get warning about rejection due to missing timing so then disabled:

Options -> Import options -> Reject Lirc Imports without timings

Then I get exceptions both from clipboard and when loading from a text file.

    Ignoring timingless remote sony_dvd_RMT-D197A in <clipboard>. Unset "Reject Lirc imports without timing" if this is not desired.
    Ignoring timingless remote sony_dvd_RMT-D197A in <clipboard>. Unset "Reject Lirc imports without timing" if this is not desired.
    Ignoring timingless remote sony_dvd_RMT-D197A in <clipboard>. Unset "Reject Lirc imports without timing" if this is not desired.
    Ignoring timingless remote sony_dvd_RMT-D197A in C:\tmp\sonydvd.lirc. Unset "Reject Lirc imports without timing" if this is not desired.

    Exception in thread "AWT-EventQueue-0" java.lang.InternalError
        at org.harctoolbox.jirc.IrRemote.toLircCodeCommand(IrRemote.java:667)
        at org.harctoolbox.jirc.IrRemote.toCommand(IrRemote.java:657)
        at org.harctoolbox.jirc.IrRemote.toRemote(IrRemote.java:643)
        at org.harctoolbox.jirc.IrRemote.newRemoteSet(IrRemote.java:160)
        at org.harctoolbox.irscrutinizer.importer.LircImporter.load(LircImporter.java:51)
        at org.harctoolbox.irscrutinizer.importer.LircImporter.load(LircImporter.java:67)
[... 117 lines cut ...]
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
        at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
        at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
        at java.awt.Container.dispatchEventImpl(Container.java:2283)
        at java.awt.Window.dispatchEventImpl(Window.java:2746)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
        at java.awt.EventQueue$4.run(EventQueue.java:733)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

clach04, 2024-03-09:
I imported one of the sony girr files, All_Sony_Codes_V1.0.girr and imported the 26.18 Pause which I *think* is one of the ones I have recorded.

When exported in lirc format there are tabs instead of spaces for indents and some key/values are present/missing in both.

Export from 2.4.1-snapshot which can be imported (generated by IrScrutinizer, then reimported):

    begin remote
    	name		remote-Sony20
    	bits		20
    	flags		SPACE_ENC|CONST_LENGTH
    	eps		30
    	aeps		100
    	zero		600	600
    	one		1200	600
    	header		2400	600
    	gap		45000
    	frequency	40000
    	begin codes
    		IrScrutinizer_scrutinized_signal	0x000000000009CB48
    	end codes
    end remote

My cut down version with tab indents:

    begin remote
    	name	sony_dvd_RMT-D197A
    	driver	devinput
    	bits	24
    	eps	30
    	aeps	100
    	one	0	0
    	zero	0	0
    	pre_data_bits	40
    	pre_data	0x4000400
    	gap	44733
    	toggle_bit_mask	0x0
    	frequency	38000
    	begin	codes
    		KEY_PAUSE	0x1A4939
    	end codes
    end remote

Which generates a different exception trace:

    Exception in thread "AWT-EventQueue-0" java.lang.InternalError
        at org.harctoolbox.jirc.IrRemote.toLircCodeCommand(IrRemote.java:667)
        at org.harctoolbox.jirc.IrRemote.toCommand(IrRemote.java:657)
        at org.harctoolbox.jirc.IrRemote.toRemote(IrRemote.java:643)
        at org.harctoolbox.jirc.IrRemote.newRemoteSet(IrRemote.java:160)
        at org.harctoolbox.irscrutinizer.importer.LircImporter.load(LircImporter.java:51)
        at org.harctoolbox.irscrutinizer.importer.LircImporter.load(LircImporter.java:57)
        at org.harctoolbox.irscrutinizer.importer.ReaderImporter.load(ReaderImporter.java:72)
        at org.harctoolbox.irscrutinizer.importer.FileImporterBean.remoteSetImporterLoadClipboard(FileImporterBean.java:174)
        at org.harctoolbox.irscrutinizer.importer.FileImporterBean.loadClipboardButtonActionPerformed(FileImporterBean.java:349)
        at org.harctoolbox.irscrutinizer.importer.FileImporterBean.access$500(FileImporterBean.java:41)
        at org.harctoolbox.irscrutinizer.importer.FileImporterBean$3.actionPerformed(FileImporterBean.java:238)
        at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
        at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
        at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
        at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
        at java.awt.Component.processMouseEvent(Component.java:6539)
        at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
        at java.awt.Component.processEvent(Component.java:6304)
        at java.awt.Container.processEvent(Container.java:2239)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
        at java.awt.Container.dispatchEventImpl(Container.java:2297)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
        at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
        at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
        at java.awt.Container.dispatchEventImpl(Container.java:2283)
        at java.awt.Window.dispatchEventImpl(Window.java:2746)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
        at java.awt.EventQueue$4.run(EventQueue.java:733)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

clach04, 2024-03-09:
Slightly cut down further:

begin remote
	name	sony_dvd_RMT-D197A
	bits	24
	eps	30
	aeps	100
	zero	0	0
	one	0	0
	gap	44733
	frequency	38000
	pre_data_bits	40
	pre_data	0x4000400
	toggle_bit_mask	0x0
	begin	codes
		KEY_PAUSE	0x1A4939
	end codes
end remote

NOTE missing:
* flags
* header

Additional entries:
* pre_data_bits
* pre_data
* toggle_bit_mask

Exception:

Exception in thread "AWT-EventQueue-0" java.lang.InternalError
	at org.harctoolbox.jirc.IrRemote.toLircCodeCommand(IrRemote.java:667)
	at org.harctoolbox.jirc.IrRemote.toCommand(IrRemote.java:657)
	at org.harctoolbox.jirc.IrRemote.toRemote(IrRemote.java:643)
	at org.harctoolbox.jirc.IrRemote.newRemoteSet(IrRemote.java:160)
	at org.harctoolbox.irscrutinizer.importer.LircImporter.load(LircImporter.java:51)
	at org.harctoolbox.irscrutinizer.importer.LircImporter.load(LircImporter.java:57)
	at org.harctoolbox.irscrutinizer.importer.ReaderImporter.load(ReaderImporter.java:72)
	at org.harctoolbox.irscrutinizer.importer.FileImporterBean.remoteSetImporterLoadClipboard(FileImporterBean.java:174)
	at org.harctoolbox.irscrutinizer.importer.FileImporterBean.loadClipboardButtonActionPerformed(FileImporterBean.java:349)
	at org.harctoolbox.irscrutinizer.importer.FileImporterBean.access$500(FileImporterBean.java:41)
	at org.harctoolbox.irscrutinizer.importer.FileImporterBean$3.actionPerformed(FileImporterBean.java:238)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

bengtmartensson, 2024-03-10:
See #521 which I think explains everything. 

> My cut down version with tab indents:
```
begin remote
	name	sony_dvd_RMT-D197A
	driver	devinput
	bits	24
	eps	30
	aeps	100
	one	0	0
	zero	0	0
	pre_data_bits	40
	pre_data	0x4000400
	gap	44733
	toggle_bit_mask	0x0
	frequency	38000
	begin	codes
		KEY_PAUSE	0x1A4939
	end codes
end remote
```

If you had fixed `one`and `zero` it may have worked.

Why are you trying to achieve? There are very few questions to which Lirc+devinput+irrecord is a good answer.

clach04, 2024-03-10:
@bengtmartensson thanks for taking the time for the detailed write-up/explanation.

What's my goal, good question :-) I barely have the basics of IR figured out, but I can record and play back using a BroadLink New RM Mini3 (https://github.com/clach04/Broadlink-e-control-db-dump) device and to some extent with some raw and also lirc command line tools. :-)  I was looking to get some "clean" recordings and then export in a format that https://flirc.tv/products/skip1s-remote-universal-remote-control could use (it **appears** to be Pronto in a json container) and potentially stretch goal to also hack up a Squeezebox hardware devices to change what it listens on (it has some [lua code with hard coded NEC hex codes](https://github.com/ralph-irving/squeezeplay/blob/5e848f1e4bc2035a76045aad48c25260ff3927dd/src/squeezeplay/share/jive/irMap_default.lua#L4)).

I think you've convinced me to complete my recorder tool I wrote for the Mini3 :-) But I may try and learn a bit more, when I get some time I see how to fix the one/zero information.

My plan is to use IrScrutinizer as a validator (and maybe help increase my knowledge). So basically I have a duct-tape project ;-)

Thanks for making this tool available!

closed 2024-03-10 by clach04